RetroPie/RetroPie-Setup #997: *.WAD files + Dinothawr game (rpi2)

Haris1977, 2015-09-05:
Is it possible to add some other *.wad games that i have (like tnt.wad, earth.wad, heretic.wad etc) in retropie? I think the libretro core exists (pr-boom core - it is the core that Doom 1 uses).

Also can you add Dinothawr game as a future update?

Thanks

BillyTPilgrim, 2015-09-17:
The only one I know of which supports all the DOOM engine games as well as Brutal Doom is ZDoom, which Brutal Doom was written for. I've compiled it and tried a binary someone else built and I can confirm it works very well, on the Pi2 at least - there's a thread on the forums about it:

http://blog.petrockblock.com/forums/topic/enhanced-doom-ports/

The only block I can see to adding it would be that there's a dependency which isn't in the standard repositories (libuv-dev) - don't know how much of a problem this would be though, and I can't speak for RetroPie on this.

Working on ZDoom on Pi:
(Ultimate) Doom
Doom II
Final Doom
No Rest For The Living (extra episode made when Doom II was released on XBLA)
Heretic
Hexen
Hexen: Deathkings Of The Dark Citadel
The Master Levels (including the ingame level select menu system someone made - needs a little tweaking of the WADs but that's true on any version)
Strife (with voices and everything!)

I've also heard from others that Brutal Doom works well.

As for Dinothawr, I've compiled it myself and it works just fine.

Haris1977, 2015-09-27:
Thanks man!

closed 2015-09-27 by Haris1977