RetroPie/RetroPie-Setup #999: [RPI2] Mame 0.78 / 0.139 builds failed

nemo93, 2015-09-06:
I've updated Retropie Setup this morning and I was about to give a try to these new (eperimental) versions to check roms compatibility, perf etc. However I just can't build these. I'll have a look later today but meanwhile I've decided to raise this ticket just in case. 

Let me enclose compilation logs to this ticket. If you need anything else please shout.

http://www.pastefile.com/hqSPOA

joolswills, 2015-09-06:
probably because I use distcc and my distcc is forced to gcc 4.8 - so they just need modifiying to force gcc 4.8 also.

joolswills, 2015-09-06:
Changed to gcc 4.8 - please try again. it should def fix 2003 building, as the compiler crashed from your logs in that one.

nemo93, 2015-09-06:
No same results. I've updated Retropie Script before launching the compilation :) While writing this I'm exploring some other options. In case you're interested here are the 2 logs after build failed:

http://www.pastefile.com/rQgPP3

joolswills, 2015-09-06:
well, in the case of the lr-mame2003, your compiler crashed. Maybe out of memory ? What system ? memory split etc.

joolswills, 2015-09-06:
the compiler crashed in both cases.

HerbFargus, 2015-09-06:
I was able to get lr-mame2003 to build on my rpi2 (after like an hour of compiling) but when I add roms to the mame-libretro folder they don't show up when I refresh emulationstation. not sure if its a conflict with lr-mame4all or what. also this line shouldn't affect things in the future should it?: https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-imame4all.sh#L39

joolswills, 2015-09-06:
well, when they are out of experimental ill have to remove that line - to be honest, its only for upgrading from 2.6 - but then too much breaks really, so I should just remove some of the backward compatibility stuff for 2.6

joolswills, 2015-09-06:
in regards to roms not showing up - check ES config perhaps?

HerbFargus, 2015-09-06:
they show up fine for mame4all and lr-imame4all.

```
  <system>
    <name>mame-libretro</name>
    <fullname/>
    <path>~/RetroPie/roms/mame-libretro</path>
    <extension> </extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ mame-libretro %ROM%</command>
    <platform>arcade</platform>
    <theme>mame</theme>
  </system>
```

missing the .zip extension. probably needs to be added int platforms.cfg?

joolswills, 2015-09-06:
it's done.

HerbFargus, 2015-09-06:
thanks, yeah it works great!

joolswills, 2015-09-06:
:)

nemo93, 2015-09-07:
sorry for not replying earlier but... I've spent the rest of the day playing to all the IREM games I was desperate to see running on pi \o/ I do confirm that Mame 0.78 works amazingly well... Not sure why the build went wrong the first time but after I removed everything in the tmp/ directory everything went well the next time. Except the issue mentioned by @HerbFargus that you already fixed!

Seriously this Mame version is so much superior to the one shipped by default in Retropie that it would make sense to simply use this version as default Mame application (at least on pi2). 

All IREM games work smoothly (except Ninja Baseball Bateman that has slowdowns but still playable) with sound, the same goes with Mortal Kombat 1/2/3 (the 1 has sound issues), Darius and Ninja Warrior works ok at full speed with sound! Konami games are just fine too. Some games also work faster than with FBA (latest experimental version you just added some days ago) like Crude Busters. 

CPS1 games just play perfectly. However I got no luck with CPS2 games (all of them) and I didn't try Neo Geo games for now. 

I've seen the wonderful job done by @HerbFargus regarding the wiki section for Mame and I'll try to update the table.

If you need me to test specific things please let me know. Let me insist this version is just awesome!!

joolswills, 2015-09-07:
Thanks for the feedback - sounds good! - as it uses its own rom folder, we can ship it alongside the current one, and people who want to use it can just use the mame-libretro romfolder.

I will get a bit more feedback from others (in case there are problems missed), but it sounds like it will do well to be included and moved out of experimental.

Thanks again.

nemo93, 2015-09-07:
You're very welcome!! oh and as @HerbFargus mentioned also the build time is rather long (1h+) on pi2 compared to other emulators. Perhaps a small notification or some words in the description as a reminder would help!

Feel free to close that one off btw.

joolswills, 2015-09-07:
Advancemame 0.93/1.2 also take a long time to build

joolswills, 2015-09-07:
i don't recommend anyone builds everything from source unless they are incredibly patient, or have a distributed compiler set up etc.

joolswills, 2015-09-07:
Will consider adding some info regarding build time.

closed 2015-09-10 by nemo93