RetroPie/RetroPie-Setup #1000: [RPI2] Mame 0.139 builds failed

nemo93, 2015-09-07:
I'm still unable to build this Mame version. I've raised the available memory to the max (gpu mem = 16) so no crash or segfault this time but the compiler throws me an error from the code as it seems and refuses to build... I also have something like 7GB of available disk space so that's ok too.

Did you successfully build this? if so could you share the binary please?

HerbFargus, 2015-09-07:
I'm attempting to compile it right now. I'll let you know how it makes out.

joolswills, 2015-09-07:
I have built it yeh - but in an in emulated chroot + distcc (so I can build it on 10 threads), but not tried building on the pi yet - will test and get back about it - might just need more available ram or something (via swap).

here's the build

http://malus.exotica.org.uk/~buzz/mame.zip

HerbFargus, 2015-09-07:
Finally finished compiling after 2+ hours. it acts as if it is opening and then it keeps opening into the rgui instead. I wonder if its just my romset- I'll try with a newer one.

HerbFargus, 2015-09-07:
ran my .163 romset through clrmamepro and it works now :) I haven't tested it super extensively but it is functioning

nemo93, 2015-09-07:
that's good news. But I'm worried that I'm still unable to compile it here :( Did you do anything specific like increasing your RAM or disk/swap space?

HerbFargus, 2015-09-07:
Nope it's just the default settings. I've never changed the swap space  or ram or GPU.

nemo93, 2015-09-07:
I really don't understand what's wrong on my pi2 then :( Let me paste the logs once again

http://www.pastefile.com/v2AJOI

I was trying to build this time with 990MB+ or RAM (gpu mem=16) but same goes with less RAM. 

I've played a bit with the binary from @joolswills (thanks!) and I do confirm it is functioning indeed. However I have hard time with the control. Only few buttons from my Xbox 360 controller are actually working (like the Mame menu and start button but nothing else). With Mame 2003 controls worked straight out of the box. On some games I can't insert coins as a small text said something about coin-lock disabled or something... 

Yet I could watch games demo and this time the CPS2 games worked. Full speed as it seems. The same goes with IREM games. Mortal Kombat launched and the sound problem I have with 2003 is gone but need to double check that too. Sega System32 games launched but are way too slow to be playable (eg. Arabian Fight).

HerbFargus, 2015-09-07:
you made sure to update your setup script first? yeah I don't  know what your issue could be.

nemo93, 2015-09-07:
yep setup script up to date. And so does my system.

nemo93, 2015-09-07:
Also not sure if I'm the only one in this situation but seems that the 2010 core is actually saving stuff in the home folder. I can see 2 of them: cfg and nvram. Not sure it is an expected behavior.

joolswills, 2015-09-08:
to rule out something on your system can you try with a fresh image perhaps ?

I can't reproduce it, but your GGC is still crashing.. 

What is the output of

```
gcc-4.8 --version
```

?

nemo93, 2015-09-10:
pi@retropie ~ $ gcc-4.8 --version
   gcc-4.8 (Raspbian 4.8.2-21~rpi3rpi1) 4.8.2

just in case:
    pi@retropie ~ $ uname -a
    Linux retropie 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux

```pi@retropie ~ $ cat /boot/config.txt
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=16
overscan_right=16
overscan_top=16
overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

# for more options see http://elinux.org/RPi_config.txt
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=384

overscan_scale=1
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2

hdmi_ignore_cec=1
dwc_otg.speed=1
```

joolswills, 2015-09-10:
maybe remove overclocking / (and set gpu_mem_1024=16 as you did before) - gcc is crashing on your setup and not on hours, but the software versions are identical.

nemo93, 2015-09-10:
I'll do that! could u share your o/c settings please or at least the values you set for arm freq etc.?

joolswills, 2015-09-10:
nothing set at all.

nemo93, 2015-09-10:
Ok after removing the 4 o/c lines I got it to build eventually \o/ I tried with both gpu_mem_1024=384 and gpu_mem_1024=16 and it builds either ways. So the issue was related to o/c settings... interesting! 

I've made further testing so with this freshly build mame2010 core and I'm mostly disappointed. Games work slower than with mame2003 but I have to emphasize that controls are really messed up. Some games do work out of the box but they are too slow to be playable (Mortal Kombat, Darius Gaiden) or they run full speed but buttons are not working (Gunforce2). I'll stick to mame2003 for now :)

Thanks a lot for support and assistance. I'll keep updating the compat' list on the wiki.

joolswills, 2015-09-10:
you may need to overvolt more - or reduce the overclock perhaps as it sounds like your overclock was not stable.

Will close this then - mame2010 will remain in experimental for the time being - thanks for feedback.

closed 2015-09-10 by joolswills