RetroPie/RetroPie-Setup #1001: OSMC Issues with compiling

MagiX13, 2015-09-09:
I have just installed OSMC (Image from 30. August) on my RPi 2 and wanted to compile some of the libretro cores to play some oldschool games.
I have encountered a few issues: 
* C++-4.7 is not in the repositories of OSMC --> Retropie-setup.sh doesn't run at all and fails to install the required resources. --> changed `__default_gcc_version="4.7"` in `scriptmodules/system.sh` to 4.8
* Compiling Retroarch has some missing dependencies. It says that `#include <EGL/eglext_brcm.h>` is not found. I tried installing the packages and compiling it as stated in the [RetroArch ReadMe Linux section](https://github.com/libretro/RetroArch#compiling-and-installing), which gives the same error (Note: The state under [dependencies](https://github.com/libretro/RetroArch#dependencies-pc) that they need `X11 headers and libs, or EGL/KMS/GBM`)
* pisnes and gpsp fails as well, this time `#include <bcm_host.h>` not found
* lr-armsnes, lr-gpsp fail as well and give multiple compilation errors... I'll try to find out more tonight

joolswills, 2015-09-09:
OSMC is not officially supported. gcc 4.8 exists in the raspberrypi repository included with raspbian.

joolswills, 2015-09-09:
as for dependencies - it's possible some modules lack some dependencies - but it works with a default raspbian set up. sounds like installing libraspberrypi-dev etc will help.

joolswills, 2015-09-09:
(have you looked into this like https://github.com/mcobit/retropie-osmc ?)

joolswills, 2015-09-10:
Correction regarding my first comment - I misread - yeh you will need to change __default_gcc_version to 4.8 for OSMC (which is based on debian jessie).

joolswills, 2015-09-10:
This is an updated script for retropie on osmc - https://github.com/mcobit/retrosmc - although with my recent retropie-setup changes, it should be possible to build some emulators from source with the retropie-setup script. Some missing dependencies have also been updated. I will fix up lr-armsnes / lr-gpsp after further testing.

closed 2015-09-10 by joolswills

MagiX13, 2015-09-11:
Alright, thank you for the Information. In the end I went on and installed raspbian as I had some other issues with OSMC as well. 
I didn't knew about retropie-osmc but knew about retrosmc, so thank you for the additional info :)
A short `lsb_release -c` checkwould suffice, I guess?