RetroPie/RetroPie-Setup #1003: [RPI2] Jessie upgrade - ES no more autostart on boot

nemo93, 2015-09-14:
I've decided to upgrade to Jessie... well for some reasons I can't explain! It went smooth but the only issue I've got far thus far is with the autostart ES on boot. I can't get it to work as before. I've seen Raspbian version is hardcoded in the retropie-setup script so I decided to force it to version "8" in scriptmodules/supplementary/autostartemustat.sh but still no luck... 

Apart from that whenever I launch ES from the command line it works very well and I've got no issue with any emulators (ok I'm using only RA cores). 

If I can do anything please give me a shout.

joolswills, 2015-09-14:
autostart won't work on jessie. it's disabled on jessie on purpose until a new method is incorporated. 

You may have collisions with jessie packages too, unless you reinstall some stuff. retropie only works on jessie since a few days, and it is currently only supported from source installs.

joolswills, 2015-09-14:
this would be best posted on the forum, as it's not really a bug - and jessie support is not "official" yet.

joolswills, 2015-09-14:
(Regarding autostarting emulationstation on systemd on jessie it's on my todo list)

joolswills, 2015-09-15:
autostart support is now working for jessie - but note jessie is not yet "recommended", and retropie is still designed primary for wheezy.

closed 2015-09-15 by joolswills