RetroPie/RetroPie-Setup #1005: Add [Theme] - Simple Big Art Theme

robertybob, 2015-09-15:
I've taken the theme from the forums by Ewzzy, made a few cosmetic changes and fixed the error which prevented me and @HerbFargus from using it (well, I've tested it on my RPi 1 anyway). 

https://github.com/robertybob/es-theme-simplebigart

HerbFargus, 2015-09-16:
I've tested it and it works now.

robertybob, 2015-09-17:
Right, should be good to go @joolswills - pushed a commit to fix another cosmetic issue. Is there anything I need to do from my end?

HerbFargus, 2015-09-18:
This one may also be worth looking at by smithers. It's basically the same thing but much more complete- though the style is more a slight alteration of the simple theme with large box art.

https://github.com/HerbFargus/es-theme-clean-look

robertybob, 2015-09-19:
Thanks @HerbFargus I'll check that one out. Have I done the right thing by creating a repo? Do I need to do some kind of pull request (i.e should I have created an issue?) to get things moving?

HerbFargus, 2015-09-20:
yeah your repo looks great, Jools will get to it when he gets to it (if you remember the last themes up on the issue tracker spent a better portion of a year or two on the issue tracker before they were included.) just be patient :) In the mean time Rookervik and I have been working on a theme, its almost complete, just missing a few controllers and needs a few minor tweaks on the logos.

https://github.com/HerbFargus/es-theme-carbon

or if you like the only box art feel:

https://github.com/HerbFargus/es-theme-carbon-centered

joolswills, 2015-09-20:
done. many thanks.

closed 2015-09-20 by joolswills

sergioad, 2016-11-25:
Somebody could help me? I am not a graphical desigber so I can not do it by miself, could you help me to add Wonderswan and Wonderswan color compatibility for simple big art? sincerely it is my favorite ES theme and I wish have those system pages themed

joolswills, 2016-11-25:
@sergioad Please use the forum for help.

sergioad, 2016-11-25:
Ok