RetroPie/RetroPie-Setup #1007: Virtual Gamepad no longer working

HerbFargus, 2015-09-17:
I wonder if this had something to do with all the Jessie stuff?

here's the errors:

```
2015-09-17 13:51:02 (2.32 MB/s) - `node_latest_armhf.deb' saved [5465884/5465884]

(Reading database ... 61209 files and directories currently installed.)
Preparing to replace node 4.0.0-1 (using node_latest_armhf.deb) ...
Unpacking replacement node ...
Setting up node (4.0.0-1) ...
Processing triggers for man-db ...
From https://github.com/miroof/node-virtual-gamepads
 * [new branch]      recalbox   -> origin/recalbox
 * [new branch]      resources  -> origin/resources
/usr/local/bin/node: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /usr/local/bin/node)
/usr/local/bin/node: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.16' not found (required by /usr/local/bin/node)
/usr/local/bin/node: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /usr/local/bin/node)
/usr/local/bin/node: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.16' not found (required by /usr/local/bin/node)
/home/pi/RetroPie-Setup/scriptmodules/supplementary/virtualgamepad.sh: line 23: pm2: command not found
/home/pi/RetroPie-Setup/scriptmodules/supplementary/virtualgamepad.sh: line 24: pm2: command not found
/home/pi/RetroPie-Setup/scriptmodules/supplementary/virtualgamepad.sh: line 25: pm2: command not found
```

joolswills, 2015-09-17:
a diff would be better so I can see the changes.

Looks like those binaries are not for our version of linux (libc) anyway - we should really be building from source etc.

joolswills, 2015-09-19:
does https://github.com/RetroPie/RetroPie-Setup/commit/1d501d9f8768574e5523975cc4ab932925325c35 help ?

HerbFargus, 2015-09-20:
This is now fixed. thanks.

closed 2015-09-20 by HerbFargus