RetroPie/RetroPie-Setup #1008: Reicast install failed

mauruco, 2015-09-18:
Hallo, I can't install reicast from retropie_setup.sh ("Conflic amr7-a").

mauruco, 2015-09-19:
The shell output is this:
../../core/hw/aica/aica_mem.cpp:1:0: warning: switch -mcpu=cortex-a7 conflicts with -march=armv7-a switch [enabled by default]
mkdir -p obj/hw/aica/
g++  -I/opt/vc/include/ -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  -I../../core -I../../core/deps -I../../core/khronos -I../linux-deps/include -g -O3  -D RELEASE -c -D TARGET_BEAGLE -D TARGET_LINUX_ARMELv7   -frename-registers -fno-strict-aliasing -fsingle-precision-constant  -ffast-math -ftree-vectorize -fprefetch-loop-arrays  -DUSE_OSS -DTARGET_RPI -DUSES_HOMEDIR -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -marm -march=armv7-a -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard -funroll-loops -fno-exceptions -fno-rtti -DARM_HARDFP -DGLES -fomit-frame-pointer ../../core/hw/aica/dsp.cpp -o obj/hw/aica/dsp.build_obj
../../core/hw/aica/dsp.cpp:1:0: warning: switch -mcpu=cortex-a7 conflicts with -march=armv7-a switch [enabled by default]
mkdir -p obj/hw/aica/
g++  -I/opt/vc/include/ -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  -I../../core -I../../core/deps -I../../core/khronos -I../linux-deps/include -g -O3  -D RELEASE -c -D TARGET_BEAGLE -D TARGET_LINUX_ARMELv7   -frename-registers -fno-strict-aliasing -fsingle-precision-constant  -ffast-math -ftree-vectorize -fprefetch-loop-arrays  -DUSE_OSS -DTARGET_RPI -DUSES_HOMEDIR -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -marm -march=armv7-a -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard -funroll-loops -fno-exceptions -fno-rtti -DARM_HARDFP -DGLES -fomit-frame-pointer ../../core/hw/aica/aica.cpp -o obj/hw/aica/aica.build_obj
../../core/hw/aica/aica.cpp:1:0: warning: switch -mcpu=cortex-a7 conflicts with -march=armv7-a switch [enabled by default]
^Cmake: *** Deleting file 'obj/hw/arm7/arm7.build_obj'
Makefile:79: recipe for target 'obj/hw/aica/aica.build_obj' failed
make: *** [obj/hw/aica/aica.build_obj] Interrupt
Makefile:79: recipe for target 'obj/hw/arm7/arm7.build_obj' failed
make: *** [obj/hw/arm7/arm7.build_obj] Interrupt

joolswills, 2015-09-19:
please post a full log via pastebin or similar.

mauruco, 2015-09-20:
This is the log. But now worked (6 attempts), I guess I should have tried harder times. Apologies and thanks for everything.

http://pastebin.com/Ews16qnL

joolswills, 2015-09-20:
not sure then - although fro that log it looks as though git may have crashed at some point when checkout out the source. Without a log of the failed build I won't be able to advise, but glad you got it working.

closed 2015-09-20 by joolswills