RetroPie/RetroPie-Setup #1009: zip support for Atari 5200, Intellivision, Neo Geo Pocket, Vectrex and Virtual Boy

ProxyCell, 2015-09-18:
Will the emulators for these systems handle zip support, can this be added in the platforms.cfg safely?

joolswills, 2015-09-18:
It might be worth me automatically adding .zip for any libretro emulators.. Since retroarch itself supports zip. there might be a few cases where zips dont work, but its easier to maintain then.

ProxyCell, 2015-09-18:
hooray! i will begin zipping up some stuff then...

gr33k, 2015-09-18:
Omg yes Please :) ZIP support for all emulators would be great to save a bit of space too.

Of course- if this doesn’t take a big performance hit...

-- 
Paul

ProxyCell, 2015-09-18:
@gr33k - well i'm not sure if all emulators support this even, i was just ask actually - specifically for the ones where just a controller can be used (not like MSX/etc)

gr33k, 2015-09-18:
@ProxyCell - Gotcha, Console based for sure. Even if it was all the RetroArch emus it would seem to be great addition.

I like the Zip for clean rom folders like instead of having Bin/cue for ISO’s for example. Otherwise the space benefit is the main perk I suppose.

Thank you folks!
-- 
Paul

joolswills, 2015-09-18:
actually I don't think atari 800 (atari 5200) emulator and jzintv (intellivision) support zips. added for neo geo pocket. because extensions and also emulationstation is all done by system, it's really down to the emulator. for some systems there are multiple emulators, some may support zip, others not.

I have added zip for some platforms that have libretro emulators that should support it. not for ones that use cd images though.

closed 2015-09-18 by joolswills