RetroPie/RetroPie-Setup #1010: RP3.0 mupen64plus PS3 Controller Bluetooth

mauruco, 2015-09-19:
Hello. The PS3 controller (bluetooth) did not work in mupen64plus. I changed the input_device on the PLAYSTATION (R) 3Controller (xxxxxxxxxxxx).cfg from "PLAYSTATION (R) 3 Controller (xx: xx: xx: xx: xx: xx)" to "PLAYSTATION(R)3Controller(00265C228505)" (without ":") and also create the InputAutoCfg.ini setting [PLAYSTATION(R)3Controller(00265C228505)].
Now it works perfect.

joolswills, 2015-09-19:
this is already fixed actually - if you remove it, update retropie-setup, and reconfigure on ES it will strip the : 

see https://github.com/RetroPie/RetroPie-Setup/commit/73538c78de12bc11f14f753232e0be74ac662d22

joolswills, 2015-09-19:
assuming you are referring to the retroarch generated joypad configs ?

mauruco, 2015-09-19:
Yes the  retroarch generated joypad configs.
My /opt/retropie/supplementary/emulationstation/scripts/configscripts/retroarch.sh looks like this:

onend_retroarch_joystick function () {
     Local device_type = $ 1
     Local device_name = $ 2
     # Sanitize filename
     Local file = "$ {device_name // [\ \ <\> \\\ /: \ * \ |] /}. cfg"
     if [[-f "/ opt / retropie / configs / all / retroarch-joypads / $ file"]]; Then
         mv "/ opt / retropie / configs / all / retroarch-joypads / $ file" "/opt/retropie/configs/all/retroarch-joypads/$file.bak"
     fi
     mv "/tmp/tempconfig.cfg" "/ opt / retropie / configs / all / retroarch-joypads / $ file"
}

And not worked. I will do the update again and test.

joolswills, 2015-09-19:
that looks correct - but you need to regenerate the configs. it should strip colons now (and other characters)

joolswills, 2015-09-19:
you can regenerate configs based on ES es_temporaryinput.cfg manually by doing

```
/opt/retropie/supplementary/emulationstation/scripts/configscripts/inputconfiguration.sh
```

from terminal.

mauruco, 2015-09-19:
I did the update again and it's all working. Sorry my mistake.I should have made an update.

joolswills, 2015-09-19:
no worries. glad it's sorted. thanks.

closed 2015-09-19 by joolswills