RetroPie/RetroPie-Setup #1011: Only Theme That Works is ColorPi v1 or v2 when you put a lot of emulators

gr33k, 2015-09-19:
Hi all,

I can reproduce on 2 Ras pi B+ so far. Load it up with a single rom per emulator for like 8+ emulators (something like that put 10+ and for sure it should happen), and Simple theme will start hanging with non loading icons for some emulators...then if you reboot, stuck at white screen.

ColorPi (v1 and v2) is the ONLY theme that seems to not be affected by this, thus leading me to believe it has something to do with themes that use different backgrounds.

I've tried with all kinds of memory splits from 32 to 512 to no avail.

Tried with RetroPie 3.0 (also the betas before) and Smither's dual boot build with the same results too, to my dismay.

I'm posting this because this may help others in the same boat to at least have a workaround until the problem is resolved. ColorPi theme works but I would love to be able to use the other themes too and/or build my own with different backgrounds.

Anyone figure out a way around this or a proposed fix?

HerbFargus, 2015-09-19:
Welcome to the processing power of the raspberry pi (or lack thereof) known issue.

http://blog.petrockblock.com/forums/topic/retropie-v3-00-rpi2-final-emulationstation-freezes-with-lot-of-systems-10/

Only solution is to remove systems or up your gpu split

gr33k, 2015-09-19:
hmmm I'm willing to up the GPU split as high as possible if it would fix it...have you or anyone got this working with a higher memory split? I will try to see how high I can take it right now and report back in a few minutes.

Removing systems isn't the desired work-around as I want to use all possible emulators (128GB stick hehe). It does work if you keep it down to a few emulators.

Nobody mentioned the Color-Pi theme is a good work around though :) I can load up as many systems as I want thus far and it's not messing up (beats no theme at least).

gr33k, 2015-09-19:
Sooooooo....raspi-config is appending the memory split after other entries which override it:

# for more options see http://elinux.org/RPi_config.txt
#gpu_mem_256=128
#gpu_mem_512=256
#gpu_mem_1024=256
overscan_scale=1
gpu_mem=896
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2

For me on the B+ it's the gpu_mem_1024=256 line that was resulting in less memory than I had thought to check (derp) thinking gpu_mem=896 would take apply if set explicitly - it does not...so I commented out the other lines.

960 GPU split is too high it seems and craps out the Pi...had to pull SD card and set config.txt back to 944. This seems to be the limit according to:
https://www.raspberrypi.org/documentation/configuration/config-txt.md
(Mental note to read documentation first then tinker lol..)

Setting to 944 and Pi is dragging along and doesn't want to fully boot up still (doesn't seem to be an efficient memory split). Wasn't patient enough to wait longer, so I set it back to 896 for further testing (again via SDCard reader). This seems to be the highest GPU split that I could do without emulation station not able to boot.

That said - All themes with multiple backgrounds seem to hang at white screen during boot.

Tested es-theme-clean-look, es-theme-paradox, es-theme-tronkyfran works fine after changing to it from colorpi theme, but startup hangs at white screen...best one being tronkyfran which was perfect and had a background for everything.

It seems like Raspi can handle all the backgrounds in some cases, but hands during boot up specifically. Additionally - it seems to run these emulators and Kodi without running in to memory issues - it seems odd to me that something like background pics would cause a memory limit on a nearly 900MB GPU memory split :(

I also stopped being lazy and counted how many emulators/icons I have on my setup (including retro pie settings, infocom,etc: 32 total.

Is this really a limitation on RasPi you think, or is there something wrong with how themes are loaded up during boot-up? If it loads it find AFTER colorpi starts up...why can't it do it at bootup with other themes? If it can handle up to 32 emulators after colorpi starts up first, why can't it boot up with like say...15 emulators?

Just seems odd! I'm happy to grab any logs or whatever I can do to help :)

Thank you

HerbFargus, 2015-09-19:
Well part of it is also the way emulationstation was coded but developments on that has halted for the foreseeable future so unless you're able to come up with a more powerful pi or a more lean theme things are kinda stuck where they're at. Also github is more meant as an issue tracker primarily for the script and emulators.  issues such as this are better addressed on the forum (of which this issue has been addressed rather exhaustively.)

gr33k, 2015-09-19:
Thanks for the quick repies HerbFargus :)

I was not aware that emulationstation development was halted! Bummmmer :( 

This is really an ideal front-end for the Pi it would seem - it's a shame. Are we looking at other front-ends to replace it maybe?

I have reposted my testing on the forum link you provided as well.

Thank you again :)

HerbFargus, 2015-09-19:
there are some other front ends some people have played with that are java based but nothing has come of them yet. I'm currently working on a theme with rookervik that should hopefully be more conducive to multiple systems but we'll see how it turns out. the points you raise are valid, we're just not quite there yet.

closed 2015-09-19 by joolswills