RetroPie/RetroPie-Setup #1013: Display setup screen when none of the previously configured controller are connected

arthurlutz, 2015-09-21:
I'm not sure this is the right place for this bug report, maybe it's an EmulationStation issue (https://github.com/Aloshi/EmulationStation/issues). 

My use case is the following : 

* start up retropie the first time with controller1 connected
* configure controller1 
* play game
* take the pi somewhere else without controller1
* boot pi with controller2 
* cannot configure controller2 since emulationstation is already configured

I had to ssh into the pi, move away /home/pi/.emulationstation to a backup and reboot the pi (there is probably a better way) to get the configuration setup at boot up. 

Then I had an issue with being able to use the controller in emulationstation but not in the games, but I'll post another issue about that. 

ps. having no USB keyboard, I wanted to use X11VNC to remotely control emulationstation via vnc from a laptop on the same network, only to realize that there was no Xorg, only framebuffer being used, do you know of a way to remotely connect to framebuffer ?

joolswills, 2015-09-21:
you don't say which version of retropie you are running - if using 3.0 image, if ES doesn't see a controller, it prompts for input entry again. Unless you have a keyboard input set up, in which case it won't prompt.

You can always have a keyboard config enabled and then navigate to configure a new controller, or just set up both controllers on ES before you leave. This isn't a bug though.

If you remove .emulationstation and then re-run emulationstation you will break the automatic controller configuration for retroarch - which sounds like what you did.

as for last question - you can't afaik. Just get a USB keyboard :)

joolswills, 2015-09-21:
You also to aid with this ticket you need to supply information.

Such as the emulationstation configs after configuring controller 1 etc.

arthurlutz, 2015-09-21:
@joolswills thanks for your reply. Indeed I did configure a keyboard on first install which explains why no configuration was offered. Can't retropie or emulationstation detect that the used keyboard is absent ? or that there is no keyboard attached ? 

I am using retropie 3.0 yes. 

Yes I guessed that removing .emulationstation would break some things, how do you unconfigure the keyboard if the design is not to offer configuration when keyboard is configured ? 

I think the use case of bringing one's pi in a pocket is nice, you generally find micro-usb, game controllers and HDMI cables at people's places, more rarely USB keyboards since laptops and tablets have become more popular that desktop computers.

joolswills, 2015-09-21:
regarding detecting keyboard - no. it can't - I assume it "may" be possible, but this isn't going to happen for your use case. Just configure both controllers before you take your machine somewhere. 

you can configure additional controllers or change your defined keyboard input by pressing start in ES.

so - Configure both controllers first - or buy a small usb / bluetooth keyboard.

closed 2015-09-21 by joolswills

arthurlutz, 2015-09-21:
As explained, a pi fits in a pocket, a small usb / bt keyboard do not. As for configuring the second controller before moving, it's not possible, since the use case is to use whatever controller is available at the destination (a friend's place, one's parents place, etc.).

Shame that this use case does not seem to be of interest to this project. Would a contribution (patches) be welcome or not ?

joolswills, 2015-09-21:
Don't configure a keyboard then, and then it will prompt to configure a gamepad when a new one is plugged in.

joolswills, 2015-09-21:
I don't see this as a problem sorry - you are the only one that has this awkward use case that revolves around your not wanting to carry a small keyboard around, or controller etc :) You can get tiny keyboard that fit in the pocket btw which would be sufficient for choosing the ES start menu to configure another controller.

If you are not going to use a keyboard, easiest thing is to not configure ES for keyboard, and then it will always prompt for a controller if not present.

arthurlutz, 2015-09-21:
About not configuring the keyboard : when using retropie to play PC games that use the keyboard it would be a shame to have to switch back to a game controller when in emulation station. I'm sorry but I don't understand this design decision. 

I don't think you have answered this question : "would a contribution (patches) be welcome or not ?"

joolswills, 2015-09-21:
You are creating this problem - by not wanting to purchase / carry a small pocket sized keyboard. I have given two solutions, but it looks like you are not wanting to compromise - we don't target the software for the wishes of a single person :)

if you can come up with a patch for our ES fork that does reliable keyboard detection (bluetooth, usb etc), then it would be considered.

joolswills, 2015-09-21:
ok. so a third solution - create a script that removed the keyboard configuration section from es_input.cfg before you take the device away with you?

Lokathor, 2016-11-28:
I also ran into this situation as well, which is how I came to this particular github issue just now. Telling someone "just buy a USB keyboard and take it everywhere you take your pi" is not very helpful in this situation.

I think that if there is a keyboard configured but no keyboard connected then it should prompt for a configuration of a new device the same as if a keyboard config didn't exist at all. I admit that I don't know the particulars of linux hardware detection, so I don't know if that's easy or not. It [seems possible](http://serverfault.com/questions/126291/ubuntu-keyboard-detection-from-bash-script) based on a quick check of stack overflow though.

joolswills, 2016-11-28:
I don't think there is a reliable way of checking if a keyboard is connected, and it's not something I want to change - the link you gave is not helpful in the case of ES. Keyboards can also be connected via ps/2 etc. Note that emulationstation is written in C++ not shell script.

The current behaviour is also used by people - making a keyboard config so it never prompts for a joypad. If no keyboard is configured it will prompt for a joypad if it doesn't detect one. If a known joypad is hotplugged at the input configuration and a button pressed it will continue. So in your case just don't configure a keyboard in ES, and it will always prompt for a controller. This also makes sense to not configure a device you don't have access to.

mrdally204, 2017-07-14:
I ran into this problem just today, I made a nes cart pi zero with retropie installed.  When I configured it I used a keyboard and then shipped it off to my buddy who order his own controller.  Long story short, the work around I came up with requires a USB thumb drive.  I had him plug it in, which prompts retropie to copy config files to the drive.  I then had him modift the es_input.cfg file and remove the block of code that had the keyboard settings.   Save that file to the to_retropie config folder, plugged it back into the pi.  After a minute he rebooted and upon plugging in his controller he was prompted!  I was totally pumped as I didn't think there was any chance to get it working without a keyboard, but a little problem solving and knowing how the system works helped me talk a novice through the process over the phone in 10 mins.  I know it's an old thread, but hope this helps.