RetroPie/RetroPie-Setup #1014: Configure libretro to use a connected controller when previously configured controllers are not available

arthurlutz, 2015-09-21:
Follow up of https://github.com/RetroPie/RetroPie-Setup/issues/1013 (again may be an EmulationStation issue)

After successfully configuring the controller in EmulationStation I started a 1 player game and it would not "see" the controller. What I guess is happening is that it is launching libretro with controller 1 configured, not controller 2. I ended up launching a helper script via ssh to configure the libreretro bindings and deleting a couple of files somewhere (sorry can't remember where or which script) to be able to play the games with controller 2. 

I guess RetroPie or EmulationStation keep an order of the configured controllers and uses that to generate libreretro configurations. While generating that configuration, it should take into account the combination of the order and the availability of the controller so that one can unplug a device and still be able to use the emulator.

joolswills, 2015-09-21:
no order configured by default - might map to which port connected etc - you can set the order in retroarch though.

what you ask is not possible currently and we have no plan to implement - for help on controller mappings etc please use the retropie forum.

arthurlutz, 2015-09-21:
the configurable order is possible in recalbox and I think it's a pretty nice feature when switching between 1/2/4 player games, and as explained in #1013 I think the portability of a retropie pi box would be a great feature (for me playing retro games is an essentially social thing).

joolswills, 2015-09-21:
recalbox has a different configuration system from us. We have a ticket open for adding additional controllers.

joolswills, 2015-09-21:
recalbox has a different configuration system from us and a differently customised emulationstation. We have a ticket open for configuring multiple players which would be used for things like pifba perhaps - https://github.com/RetroPie/EmulationStation/issues/4 

Your original problem in this ticket is related to removal of the initial ES configuration which breaks configuration of retroarch, so I will close this ticket. thanks.

closed 2015-09-21 by joolswills