RetroPie/RetroPie-Setup #1015: Configure RetroPie with some touchscreen phone controllers

arthurlutz, 2015-09-21:
It would be great to be able to take your retropie box anywhere, connect it to a screen, charge it up and be able to play with the controllers that you find (see #1013 and #1014 for start of discussion). 

In that frame of mind, a lot of people have smart phones that could be used as game controllers.  A number of projects do that with PCs in mind, maybe some could be adapted to work with retropie. Ideally one could choose to use the wifi or bluetooth so that the user has the most choice in setup types (RJ45 cable on same network as wifi, add WIFI dongle, add bluetooth dongle, etc). 

A number of projects do this for example on android https://play.google.com/store/search?q=game%20controller or PC with http://gameplay3.gestureworks.com/ (see search https://www.google.fr/search?q=use+phone+as+game+controller). I haven't researched fully (see http://android.stackexchange.com/questions/22727/can-i-use-my-phone-as-a-bluetooth-game-controller-for-my-computer#25976 for a number of solutions) but some open source solution could be re-used for this.

joolswills, 2015-09-21:
https://github.com/RetroPie/RetroPie-Setup/wiki/Virtual-Gamepad

Please do check the wiki / forum before opening tickets. The forum is a better place to start.

joolswills, 2015-09-21:
Note this is implemented as a retropie module under the experimental menu.

arthurlutz, 2015-09-21:
@joolswills great news, thanks for the link.

closed 2015-09-21 by arthurlutz