RetroPie/RetroPie-Setup #1016: "Value too large for defined data type"

eusi, 2015-09-21:
Hey.

As the title says, it's not able to run a huge image (above 2gb).

In my case I tried a psx game (pbp file of 2.2gb) on pcsx_rearmed-emulator on RetroPie.

Probably issue and solution: https://www.gnu.org/software/coreutils/faq/coreutils-faq.html#Value-too-large-for-defined-data-type

Is this problem made by RetroPie or its emulator (here: pcsx_rearmed) and fixable?

Greetz
eusi

joolswills, 2015-09-21:
I don't think we are doing anything special/different in the way we are compiling, so I guess it needs fixing upstream.

adrianmoisey, 2015-09-22:
This might be because the ARM devices on the Pi are 32bit. I imagine upstream won't/can't fix this.

joolswills, 2015-09-25:
Should be able to support large files on the rpi1 with source changes though despite being 32bit.

joolswills, 2015-09-25:
Closing here as I see you have reporting it upstream. thanks.

closed 2015-09-25 by joolswills

adrianmoisey, 2015-09-25:
Do you have a link to the upstream bug? I'd like to follow it.

joolswills, 2015-09-25:
https://github.com/notaz/pcsx_rearmed/issues/59

johnweland, 2016-03-13:
notaz has said that the issues has been pushed. Not sure when retro pie will get it though.

joolswills, 2016-03-13:
you can install from source now and test it

johnweland, 2016-03-14:
I'm still kinda of new to that process, but I'll attempt to give it a go this week.

johnweland, 2016-04-26:
Hey works! I went to the experimental packages and installed pcsx_rearmed large file support is enabled. Of course I have to remap my keys and get my psx bios dropped in.

dankcushions, 2016-04-27:
the standalone fix was merged in to the libretro version a few days again. I can run 2GB .pbps on it. try recompiling that from source and it should be fine :)

johnweland, 2016-04-29:
Hey thanks, worked like a charm.