RetroPie/RetroPie-Setup #1017: Installing RetroPie non-interactively

adrianmoisey, 2015-09-22:
Is it possible to install RetroPie using the RetroPie-Setup script non-interactively?
I'd like to be able to automate an install using script.

I'd expect something like this to work:
```
./retropie_setup.sh binary
```
For binary installation.

Or:
```
./retropie_setup.sh source 105 124 209
```
To install `dgen`, `retroarch` and `lr-fceumm`.

adrianmoisey, 2015-09-22:
Is this what I want? https://github.com/RetroPie/RetroPie-Setup/wiki/Building-RetroPie-Archives

joolswills, 2015-09-22:
please use the forum for non issue questions. That page described a few things. you can get info by running

```
sudo ./retropie_packages.sh
```

which documents most calls. non experimental emulators have an install_bin function too.

to install dgen retroarch and lr-fceumm from binary and configure

```
for pkg in dgen retroarch lr-fceumm; do sudo echo ./retropie_packages.sh $pkg install_bin; sudo ./retropie_packages.sh $pkg configure; done
```

some modules may open dialogs - to make them just output to console you can add __nodialog=1

```
sudo __nodialog=1 ./retropie_packages.sh setup binaries
```

would do a full binary install / configure of everything - same as choosing option 1 in retropie-setup - but without opening any dialogs at the end.

adrianmoisey, 2015-09-22:
> please use the forum for support questions.

You should add a `CONTRIBUTING.md` to the repo to inform anyone making a PR or issue of that: https://github.com/blog/1184-contributing-guidelines

Thanks for all the info, I think it's exactly what I need. I'll play around and ask in the forum if I need more info.

closed 2015-09-22 by adrianmoisey

joolswills, 2015-09-22:
Will do thanks.

joolswills, 2015-09-22:
I accidentally put an echo (was testing) in an example script above. fixed now.

adrianmoisey, 2015-09-22:
Cool, no problem. Based on a quick look, I think this will work for me. Thanks a lot!