RetroPie/RetroPie-Setup #1018: C64 (vice) no Restore keypress?

Haris1977, 2015-09-22:
Has anyone managed to use the restore key?

PgUp doesnt work for me.

I see in /opt/retropie/emulators/vice/lib/vice/C64 there are some *.vkm files like sdl_sym.vkm (where vice keyboard mapping is hardcoded i suppose?). But there is not a restore button. I tried changing the backspace to map it to restore like this

from: 8 0 0 8 /* Backspace -> Del */
to: 8 0 0 8 /* Backspace -> Restore */

but didn’t work. Also tried to add in # Restore key mappings the following (there are null values)

280 -3 0 /* Backspace -> RESTORE */
0 -3 1 /* Backspace -> RESTORE */

Didn’t work either. Any thoughts?

Haris1977, 2015-09-25:
Anyone?

joolswills, 2015-09-25:
There is no need to "bump" tickets - please be patient and people will help when they have a moment - this support request would be best posted to the forum not the bugtracker .

joolswills, 2015-09-25:
There is no need to modify any keymaps - page up works correctly as the "RESTORE" key.

joolswills, 2015-09-25:
For support in using the emulator - please post on the retropie forum. http://blog.petrockblock.com/retropie/

closed 2015-09-25 by joolswills

Haris1977, 2015-09-25:
Have already posted here:

http://blog.petrockblock.com/forums/topic/c64-vice-no-restore-key/#post-106850

I really think that it is an emulator limitation though:(

joolswills, 2015-09-25:
Yes - I tested it and restore worked. Ill comment on the forum thread.

Haris1977, 2015-09-25:
So it must be my keyboard's layout. Thanks!