RetroPie/RetroPie-Setup #1020: lr-mame2003 - sed: -e expression #1, char 28: unterminated `s' command

snigel, 2015-09-22:
sudo ./retropie_packages.sh 218 sources
= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-mame2003' : Arcade emu - MAME 0.78 port for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/libretro/mame2003-libretro.git" "/home/pi/RetroPie-Setup/tmp/build/lr-mame2003"
Cloning into '/home/pi/RetroPie-Setup/tmp/build/lr-mame2003'...
remote: Counting objects: 2445, done.
remote: Compressing objects: 100% (2369/2369), done.
remote: Total 2445 (delta 76), reused 2309 (delta 70), pack-reused 0
Receiving objects: 100% (2445/2445), 10.02 MiB | 3.14 MiB/s, done.
Resolving deltas: 100% (76/76), done.
Checking out files: 100% (2387/2387), done.
sed: -e expression #1, char 28: unterminated `s' command

Tele42, 2015-09-22:
sed is at https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-mame2003.sh#L21, same syntax at https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-mame2010.sh#L21

joolswills, 2015-09-22:
Yeh look like a c&p fail there. will fix thanks! :)

snigel, 2015-09-23:
Could this have been avoided by using set -e in the scripts?

joolswills, 2015-09-23:
That could affect other things that return a non 0 code, yet are not errors etc - such as grep. Lots of stuff would need to change I guess, so I'd rather keep it as it is now. Also in the case that one module fails (such as a change in a remote git repository) it is better if things continue on etc.

joolswills, 2015-09-23:
(I do run the scripts though shellcheck now and again but had not in this case - the sed fail was actually harmless as it shouldn't have been there at all). Is sorted now. thanks.

closed 2015-09-23 by joolswills