RetroPie/RetroPie-Setup #1021: ES-Carbon Theme Final Review

HerbFargus, 2015-09-23:
@petrockblog @joolswills Just opening a ticket for you to review the Carbon theme for the next point release.

https://github.com/HerbFargus/es-theme-carbon

Can also be installed from the setup script.

joolswills, 2015-09-23:
I am in support of this theme - we have a lot of tickets due to people being out of gpu allocated memory due to the backgrounds etc in the simple theme - the simple theme is pretty - but requires ram. 

The carbon theme has the retropie branding, but is also elegant and stylish in a way that is memory efficient - so I would back using it as the default. :) Basically a static background, with a similar horizontal nav, but with very good looking vector images reflected above the navigation per platform.

+1 from me.

gr33k, 2015-09-23:
+1 from me - Works fine on my RasPi B+ with 32 icons - nothing missing :)

I tried the carbon and the carbon-centered versions. Both worked fine, but I opt for the standard Carbon theme as default candidate.

I DO love the Simple Theme - but I can’t use it on my Pi. So 100% agree with the static background.

-- 
Paul

Haris1977, 2015-09-24:
+1

It should be the default theme :)

petrockblog, 2015-09-24:
This theme looks awesome and the technical advantages that Jools also pointed out above make it a great candidate as new default theme.

+1 from me as well.

joolswills, 2015-09-24:
ok so all decided - we will switch to the theme (for new installs). Thanks everyone :) Will get this sorted over the next few days.

HerbFargus, 2015-10-03:
as of commit https://github.com/RetroPie/RetroPie-Setup/commit/05a665ddf49cb470084f54af850d3bf47f293121 we can consider this resolved. Thanks

closed 2015-10-03 by HerbFargus

Haris1977, 2015-11-14:
Can anyone also create a "steam" theme folder (for carbon theme) ? Check out my edit 2 in my 2nd post here for more details about that:

https://github.com/RetroPie/RetroPie-Setup/issues/1082#issuecomment-156080005