RetroPie/RetroPie-Setup #1022: No mouse support for light gun or mouse games?

davidj1987, 2015-09-24:
The emulator in question is Genesis Plus GX (Retroarch) and I can use the mouse buttons to fire, however I cannot move the cursor, is mouse support not fully implemented? Doesn't matter if it is Menacer (Genesis), or Light Phaser (Master System) games. I've tried other games that use mouse controllers and again no dice. I have used this core on the PC and it works fine with the mouse there.

I have enabled the device (menacer, light phaser, sega mouse etc) in the Retroarch GUI when I load the game (or restart) but the mouse does work in the GUI. Am I missing something?

robertybob, 2015-09-25:
Maybe try asking here if it's an emulator issue:

https://github.com/libretro/Genesis-Plus-GX

davidj1987, 2015-09-26:
I asked here, and here is the response:

https://github.com/libretro/Genesis-Plus-GX/issues/31

davidj1987, 2015-09-29:
Sorry to double post - I tried another mouse, and the same thing - the mouse works fine in the GUI but not in the emulator and I posted the same issue on the emulator authors git.

joolswills, 2015-10-15:
https://github.com/libretro/RetroArch/issues/2192

Seems to be an upstream issue with udev driver on retroarch.

cmitu, 2023-11-24:
Closing, upstream has now support for lightgun/mice 
 - https://docs.libretro.com/library/genesis_plus_gx/#input
 - https://docs.libretro.com/library/genesis_plus_gx/#lightgun

closed 2023-11-24 by cmitu