RetroPie/RetroPie-Setup #1023: advmame 1.2 not supporting more then 9 buttons

NeoID, 2015-09-25:
Seems like avdmame 1.2 is not supporting more then 9 buttons or doesn't play nice with the Xin Mo controller. It recognized all buttons for player 1, but when trying to map player 2 I can only map the first button and the rest won't get recognized in the mapping process. Anyone had the same issue?

I think it might be related to:
http://piplay.org/forum/discussion/1451/joysticksbuttons-not-responding-in-games-maybe-xin-mo-troubles

joolswills, 2015-09-25:
Can't really help with this as I don't have the hardware to test - you could try asking on the retropie forum to see if anyone there has a solution - more likely to get advice there.

You also might want to try the recently added libretro mame2003 module which is based on 0.78 mame and may work better for you (Not it has some problems with some games and inserting coins where select is the hotkey - choosing another hotkey solves this)

NeoID, 2015-09-25:
Found the solution: nano /boot/cmdline.txt
Add "usbhid.quirks=0x16c0:0x05e1:0x040" at the end of the string. This will split js0 into js0 and js1 so the device isn't presented as one single device anymore. :)

closed 2015-09-25 by NeoID

joolswills, 2015-09-25:
aah it's that issue again - thought it was something else - yeh there is a very long thread about this on the forum - glad you have it sorted.