RetroPie/RetroPie-Setup #1024: Uae4ALL keyboard remapping?

Haris1977, 2015-09-28:
I have problems with specific games - like North And South, but i assume there are plenty of others - where the right shift key press (RSHIFT) doesn’t work (with the right shift you choose which unit to move in battlefield)

I know it is a key conflict as i had the same prob in WinUae where i changed the key combination from keyboard layout B to A and worked fine.

Btw in WinUae:

keyboard layout B = cursor keys, right cntrl and ALT=fire
keyboard layout A = numeric keypad, 0 and 5=fire (even though arrow keys still work in some games)

Anyone knows if i can somehow remap my keyboard?

Have opened a thread here:

http://blog.petrockblock.com/forums/topic/uae4all-keyboard-remapping/

joolswills, 2015-09-28:
it's hardcoded, and it unlikely to be changed in our uae4all2.

Haris1977, 2015-09-28:
Thus means that many games are un-playable?:(

amigagamer, 2015-09-28:
Slightly off topic as this post relates to uae4ARM (very similar code so applies to uae4all2 as well). I did a quick hack some months back to get normal use of ctrl/shift/alt keys back for pinball fantasies etc. Both emulators use up these keys for menu access, toggling mouse mode and joystick button emulation.  My "solution" was to hardcode the menus to F11 and F12 
https://github.com/amigagamer/uae4arm-rpi/commit/d98851fe3e5fd534379f84e406b22d2aee692708
https://github.com/amigagamer/uae4arm-rpi/commit/0e08d4c5ffaa2d5d8ac3c76b9826965468492ada

and sdl joystick onto neumeric keypad.  
https://github.com/amigagamer/uae4arm-rpi/commit/9f6498c4148607df9f8848b6a1ad70f0863b74a9

I then did the following to return the keys to a normal state
https://github.com/amigagamer/uae4arm-rpi/commit/1584fc3339f0c7abba655bd4fe7d77f2a42bba24

Not a very clean job on that last commit, it breaks joystick emulation on the keyboard - works for me though as i use a usb gamepad.
Just posting this as it may help others see where the changes need to be made.

Haris1977, 2015-09-28:
So you can only use gamepad (for games) rather that keyboard with this method?

joolswills, 2015-09-28:
is this post about uae4arm ? I assumed it was about uae4all2 as the OP never specified.

joolswills, 2015-09-28:
Sorry misread.

Haris1977, 2015-09-29:
I think the best solution in here would be to add different keyboard layouts (inside the uae4all gui).

Smth like keyboard layout A and keyboard layout B, where:

keyboard layout A = numeric keypad, 0 and 5=fire 
keyboard layout B = cursor keys, right cntrl and ALT=fire

joolswills, 2015-10-15:
I doubt this is going to happen any time soon - but in the meantime, you can try out uae4arm (currently in experimental still), which does have input remapping etc.

joolswills, 2016-03-26:
As it is doubtful any further changes will be made to uae4all2, have you tried using uae4arm for this ?

vretro, 2016-03-28:
The custom key/button mapping feature of UAE4ARM is not functioning, but would be so useful if it did.

joolswills, 2016-03-28:
We are not responsible for development of all the emulators we ship with - you could try reporting it to the repository for that particular emulator - https://github.com/Chips-fr/uae4arm-rpi/

vretro, 2016-03-29:
I was replying to your previous comment as you asked, because I had tried custom key and button mapping in both emulators. I will report the issue over at: https://github.com/Chips-fr/uae4arm-rpi/

Guitarans, 2016-06-14:
HI Haris1977.
Im standing in the same situation :-( 
North and south is just So so fun and one of a kind game !!! I realy love it..
ive tryed to see if i could edit something in the uae4arm code, but it didnt work out for me. and the uae4all is hardcoded ( and hardcoded wrong in my opinion )
Well i have some good news if you are using the retro pie and need a North and south fix.
The ATARI ST emulator runs the game very well! the sounds are not exactly the same but its okay..
hope that it helps.

joolswills, 2017-04-04:
Emulator no longer developed / improvements can be fed back to the more active amiberry / uae4arm projects.

closed 2017-04-04 by joolswills