RetroPie/RetroPie-Setup #1026: Very unique keyboard issue - can't find answers after days of searching

pancakeykakes, 2015-09-29:
Alright, so I should preface by saying I have RP and Kodi installed as standalone environments on top of Raspbian Wheezy and I love it.
I have done a LOT of research, and have only found solutions for those whose keyboard does not work AT ALL within RetroPie/Emulationstation.

The issue I'm running into, is once I launch RP-- they keyboard DOES work within it, but after quitting, my keyboard will not work within the terminal. If I then launch Kodi, I can use it within Kodi. Though it still does not work within the bash terminal after. Only fix I could find up until last night is rebooting.

Last night, however, I noticed that if I use the RetroPie setup-script to enable auto-running RP at boot, I do not have loss of keyboard functionality upon exiting. Even if I relaunch by typing 
"$ emulationstation". If I disable auto-launch at boot, the problem returns.

I have been working on a little script (I'm a beginner, but I'm making good progress) for those who have a 3-way environment such as mine that also co-exists with a very unique xboxdrv setup that allows for full navigation of the three environments with no need for a keyboard. However, I would like the users to be able to use their keyboard in the terminal in case they want/need to manipulate their system in any way.

Any input (I know, bad pun) would be appreciated.

pancakeykakes, 2015-09-29:
Closing issue - found out it was an issue with xboxdrv and the order I presented arguments. My apologies.

pancakeykakes, 2015-09-29:
I have another piece of useful information.
If I use a script to kill Emulationstation, the keyboard stops working in bash. But if I use the Quit option from the Emulationstation GUI, it does not happen.

I will just remove the killswitch script I've been using and leave via GUI.

joolswills, 2015-09-30:
sounds hardware related. Also check PSU etc - maybe under load your machine is not getting enough power. Questions like this are best posted to the forum. Thanks.

closed 2015-09-30 by joolswills

pancakeykakes, 2015-09-30:
Well it only happens when killing Emulationstation by command
sudo killall Emulationstation

So I would have to disagree.
Also, power is not an issue. Brand new USB power cable and brand new 2.1A adapter, Pi is overclocked to RPI2 preset.

joolswills, 2015-09-30:
Then I have no idea - you are the first one to report such an issue - I have used a few different keyboards and have not come across this. Maybe it's something specific to your setup/config - can you reproduce it from our prebuilt image ?

joolswills, 2015-09-30:
it might be that killing it like that is leaving the framebuffer in an unusable state, which could look like the keyboard isn't working.

joolswills, 2015-09-30:
are you launching ES from within Kodi or ? There are problems with the framebuffer being left in an unusable state after exiting the old version of Kodi - see http://blog.petrockblock.com/forums/topic/exiting-kodi-to-emulationstation/

The framebuffer can be left in a bad state when killing other apps too.

I can't really advise further. It's not a retropie bug though.

pancakeykakes, 2015-09-30:
I am killing Emulationstation via a script bound to a controller. The script is literally just the command listed above.

The framebuffer idea is something I haven't thought of. If you have any documentation on that stashed away somewhere I would love a link.

That actually leads me to agree with you that RetroPie is not to blame.

joolswills, 2015-09-30:
you can try doing "fbset -depth 8 && fbset -depth 16" to see if it restores it in your script.

more info on that thread I linked - and maybe check the raspberrypi forums.

joolswills, 2015-09-30:
Also instead of SIGTERM which is default did you try a killall -SIGINT emulationstation ?

pancakeykakes, 2015-09-30:
I will try both of these ideas later on today. Past that I will head back to Google. 

Thanks for the info!

pancakeykakes, 2015-09-30:
Your fbset depth issue resolved the bug, thank you!

Tele42, 2015-09-30:
FWIW, I have also had lost keyboard input in the VT after some emulators crashed out, I was still able to interact with emulationstation and the emulator, but I could not enter the text menu to change which emulator to use. Keyboard input always worked fine on first run. Once it stopped accepting input on the VT, nothing helped except a reboot.