RetroPie/RetroPie-Setup #1027: Direct link to ROMs with game's name

mwithi, 2015-09-30:
Hi! At the moment RetroPie shows in the Amiga games list only the scripts to launch the emulator.

At the moment I have:
+Start UAE4All
+Start UAE4All2

Anyone know how to have instead scripts to launch the games directly?

BillyTPilgrim, 2015-09-30:
I replied to a thread on this subject a few days ago here:

http://blog.petrockblock.com/forums/topic/amiga-via-emulationstation/#post-106262

Hopefully that'll be of some help.

closed 2015-09-30 by joolswills

mwithi, 2015-09-30:
Thank you Billy, I'm trying that script but I'm trying also to adapt to work with Uae4All2 and I just get:
tvservice-client: Failed to connect to TV service: -1

Does the solution work only with Uae4Arm?