RetroPie/RetroPie-Setup #1029: Official support for Raspbian Jessie

joolswills, 2015-10-02:
Jessie is now released on the raspberrypi.org site. Need to build and test new images based on Jessie and build binaries for Jessie also.

https://www.raspberrypi.org/blog/raspbian-jessie-is-here/

pancakeykakes, 2015-10-02:
So far I am not impressed with Jessie. Based on what I read, that is.

joolswills, 2015-10-02:
Please do share any links you think would be useful. We don't include desktop components in our prebuilt images, but we would benefit from updated versions of some userland components.

HerbFargus, 2015-10-02:
So is it your intention to have 3.1 be based on Raspbian Jessie- or is this for future releases that you're looking at? I'm happy to give any new images a test

joolswills, 2015-10-02:
An undecided future release.. some time after 3.1

pancakeykakes, 2015-10-02:
How do you guys feel about making a standalone version for Jessie? That would make me a bit more tempted to try it out. I run Wheezy with Kodi and RetroPie on top of it as standalone environments.

Another thing is if you guys do plan on that, I will be able to update my TriPi script for controller navigation into it for any of my users that want to upgrade.

joolswills, 2015-10-02:
At some point I guess there will be a Jessie based image for testing as well as the standard version, but I am not going to build 2 sets of images for every release.

pancakeykakes, 2015-10-02:
Fair enough.

lukeroge, 2015-10-03:
Another +1 for Jessie support. I moved my multi-use project to Jessie to get newer python versions, and official support for on-the-side emulation would be nice.

joolswills, 2015-10-03:
Note you can install it manually on Jessie now - just no binary installs available, but if you wanted to build some emulators from source that should all work.

lukeroge, 2015-10-04:
I had some bizarre issues installing emulationstation, but I think it might just be my image corrupting. Apt kept locking up on installing libboost, despite trying all kinds of dpkg and aptitide tricks. Oh well! I'll update my post after a reimage.

diogomsantos, 2015-10-07:
I just tried making a source based instalattion in a clean official Raspbian Jessie Image and the first error I got is building SDL1. Here is the log:
http://pastebin.com/Uf4g1VuJ

Now it started compiling the emulators and will let it finish, meanwhile if I found another error I will paste it here.

joolswills, 2015-10-07:
thanks will check/debug - this is new - I guess they added a prepatched rpi sdl to jessie now that it's officially released (rpi foundation are using our exiting patchsets, so I need to check for that).

joolswills, 2015-10-07:
or just forcing a vanilla debian sdl would be fine. We could just accept the raspbian sdl as it's already got our fixes in, but we might add new stuff later (perhaps).

diogomsantos, 2015-10-08:
Ok, I already had SDL 1.2.15-10 and SDL 2.0.2 installed from the Raspbian repo, but wasn't sure that will work. 
Everything seemed to compilied fine including SDL2, unless lr-mednafen-wsan and lr-mednafen-ngp which failed because /home/pi/RetroPie/tmp/build/$emulator/README.md not found in both.

Will test now ES and some emulators to check if they are working.

joolswills, 2015-10-08:
Sounds like you need to update retropie-setup - lr-mednafen-wsan and lr-mednafen-ngp were fixed up a week ago.

joolswills, 2015-10-08:
5 days not a week. sorry.

joolswills, 2015-10-08:
Please can you post me more of the log from earlier - It's not complete and I need to see the source stage of the sdl1 build/fail (As It worked here)

joolswills, 2015-10-08:
I can probably reproduce it from a fresh image, but will save me time - I'm assuming there is a difference between our deb-src apt sources. Downloading a vanilla Raspbian jessie image now anyway to see.

joolswills, 2015-10-08:
should be fixed now.

diogomsantos, 2015-10-08:
Yes I had Raspbian Jessie installed about a week ago and cloned RetroPie-Setup +/- at the same time but only now I had the opportunity to lay hands on that. Updating RetroPie-Setup fixed ngp and wswan.

The old log was corrupted (sometimes logs get corrupted, don't know why but that already happens in Wheezy). Anyway here is a new one:
http://pastebin.com/cms63qMP

Another thing I noticed that doesn't work is the splashscreen. I tried multiple, the sysvinit script is processed but nothing happens. I believe that's because of systemd. I already tried to make a systemd service for fbi but still not working yet (exits with "Oops Terminated).

joolswills, 2015-10-08:
I have the splashscreen working ok here on jessie - so not sure what the problem is. I'll try from a fresh image though - you did "enable splashscreen" from the configuration ?

joolswills, 2015-10-08:
Tested on a fresh Raspbian jessie and it worked fine - did you configure it to boot to "Console Autologin" via raspi-config?

Without that it will boot to a desktop, and the desktop starts loading early enough as to not give the splash a chance to startup.

diogomsantos, 2015-10-08:
Sorry didn't saw your message about SDL1, it compiles fine now. Splashscreen also working just fine, the problem was that I was using an old custom one, thanks.

Some things I noticed that the source-based installation doesn't install,  don't know if this is expected or not:
-es-theme (none are installed)
-retropie menu
-samba shares  (not sure about this)
-ES autostart

ES (from terminal and autostart enable/disable) is working fine, USB gamepad configuration in ES also, but I didn't test the emulators yet. 

It seems that all RetroPie-Setup scripts are working fine but I didn't test some because I don't have the hardware to test, like:
-wireless/bluetooth gamepads like PS3 and XBOX 360
-xarcade2joystick

RetroPie Menu options are working also.

Maybe it will not be needed much work afterall to get RetroPie working fine in Raspbian Jessie :D

Time to test the emulators with one USB Logitech gamepad.

joolswills, 2015-10-08:
es-theme - you are installing from source - a default theme is only set up when installing from binaries, but you can just install a theme via the supplementary menu - there is a theme installer. 

retropie menu is expected to be installed manually if using the source install, same for shares, and ES autostart. they are configured from the supplementary menu.

There was plenty of work to get RetroPie-Setup working in Jessie :) - The work was done some weeks ago. This ticket was opened for me to track "official" support - ie, out of the box with binaries and shipping jessie by default.

diogomsantos, 2015-10-08:
Yes I know that options can be installed from the RetroPie-Setup script, I have done that already.

So maybe the next step is to test the binaries compiled from source, and if they are working, to make binary packages, put them on the RetroPie server and enable the option to Install binaries.

joolswills, 2015-10-08:
It's already in progress - this was why I opened the ticket for myself :)

HerbFargus, 2015-10-08:
So will there be two binary options then when updating? One for wheezy and one for Jessie?

joolswills, 2015-10-08:
Just one. It will know which distro you are on.

joolswills, 2015-10-08:
I actually built binaries for Jessie a week ago, but then got busy doing the 3.1 release - I'm building them now again though.

HerbFargus, 2015-10-09:
Just as a reference for others: this is how I installed retropie on raspbian jessie manually (from source):

```
Installing Retropie Manually (from Raspbian Jessie):

INSTALL RASPBIAN:

https://www.raspberrypi.org/downloads/raspbian/

unzip
write to sd card with win32diskimager

sudo raspi-config

boot options
console autologin

expand filesystem
reboot

INSTALL RETROPIE:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git

cd
git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git

cd RetroPie-Setup
sudo ./retropie-setup

Install Emulationstation

Auto Start Emulationstation

Install Theme from setup script

Enable Samba Shares
(hostname is raspberrypi not RETROPIE)

Install Retroarch

Install your emulators

transfer your roms

reboot
```

HerbFargus, 2015-10-10:
just ran a full binary install: 

could not install package(s): libxaw7-dev checkinstall.

full log:

http://pastebin.com/8ZhjQ9u6

may have been a network thing again but  not necessarily sure- it all seems to work though, still testing

joolswills, 2015-10-10:
Your install issue was a network problem. 

Will look into mgba when i get a chance. thanks.

HerbFargus, 2015-10-10:
yeah lr-mgba issues are specific to jessie as compiling from source on my wheezy build had no issues.

joolswills, 2015-10-10:
you could try compiling just that emulator from source on jessie - might be my cross compiler.

HerbFargus, 2015-10-10:
thats the next thing on my list- will report results

HerbFargus, 2015-10-10:
same issue:

here's the build log:

http://pastebin.com/9ceKhY8C

joolswills, 2015-10-10:
ok thanks.

carlosouro, 2015-10-11:
Hi guys,

PS3 Controller Setup is not working in Jessie.
No expert here, but it seems like it's due to how bluetooth support has changed in Jessie.

bluez-utils and other packages no longer exist in Jessie (apt-get says it has been replaced with just bluez), making #includes fail during setup.
Then the ps3 setup screen comes up but fails saying no PS3 USB controller found and it fails all together.

Errors given at the end:

* Could not install: bluez-utils bluez-compat bluez-hcidump libusb-dev libbluetooth-dev joystick libusb-1.0-0-dev
* Could not successfully install (/home/pi/RetroPie-Setup/tmp/build/ps3controller/sixpair not found)
* please see log blah blah (let me know if you really really need it)

joolswills, 2015-10-11:
ps3controller should be fixed now.

joolswills, 2015-10-14:
the mgba problem should be fixed. I'm going to build new binaries for wheezy / jessie now.

ProxyCell, 2015-11-12:
Hi,

Is building from source on Jessie working for anybody? I just ran my first one and it threw a ton of warnings/errors/failures at me. I did try to built a lot of stuff at once and my next run will be a bit easier but I was wondering if you guys had any experience with it already.

HerbFargus, 2015-11-12:
I installed it manually piece by piece and ran into a few dependency errors but they were fixed for the things i tried. I didn't try a whole lot of things though. I did the partial install explained on this page:

https://github.com/RetroPie/RetroPie-Setup/wiki/Manual-Installation

joolswills, 2015-11-12:
@ProxyCell without any log I can't advise.

joolswills, 2015-11-12:
I would add that source builds can break often due to upstream changes. Stuff is fixed as/when I have a chance. Binaries are available for Jessie, so I would use those.

ProxyCell, 2015-11-12:
@HerbFargus & @joolswills 
Thanks a lot for the replies! I will begin the next run tomorrow and I will do them one by one.

jimbo27, 2015-12-08:
Hi guys,

after >1 year of using Retriopie an a Pi1 I decided to step over to a Pi2. For a fresh new install I decided on Jessie because of its future compatibility and perhaps better support for Pi2.

I installed Jessie Lite (from https://www.raspberrypi.org/downloads/raspbian/, version November 2015).

Following the hints above I tried to install RetroPie step by step (partial install). I began with git, dialog and then the RetroPie script. After that I tried to install Emulationstation through the RetroPie-Frontend. The binary version failed to install so I tried the source version which installed almost error free (I saw some dependency problems in the log). Anyway: After installing the simple theme Emulationstation started successfully on my Pi2. es_systems.cfg were missing so I had to get an actual version from the internet.

After that I wanted to see my first emulator running on my RetroPie system. I decided on Retroarch and Libretro cores for SNES. I started with Retroarch. This time I could not compile the source version so I installed the binaries. After that I installed two Libretro cores (Pocketsnes and SNES9x and some roms.
Unfortunately Emulationstation could not start the emulator. runcommand.log says:
/opt/retropie/emulators/retroarch/bin/retroarch: error while loading shared libraries: libjack.so.0: cannot open shared object file: No such file or directory

So my installation seems to be broken. Any hint what I did wrong?

PS: I am interested to get this thing working on Jessie. I am no coder but I am ready to help with log files of install attempts of binary or source versions. I am capable of making backups anytime in order to wind back to a previous version of my system. For example I could go back to a clean system without RetroPie. I could try to install every component step by step and post the resulting logfiles. Would this be helpful?

HerbFargus, 2015-12-08:
RetroArch is currently under a lot of development so its more than likely it will break from source than it will build, so I would stick with binaries for everything (if they were built for jessie) as there really isnt a whole lot of benefit to building from source.  We've been sorting out 3.3 pending RetroArch and holidays and then perhaps we'll focus more on Jessie support. Ideally RetroPie should be modular enough to build from individual pieces but sometimes dependencies slip through (this will be sorted in time) 

As far as what you are doing, if you do have logs you can put them on pastebin and post links for your build logs so we know what broke it. @joolswills I know this isn't top priority right now but for long term do you think it would be worth making a chart on a google sheet or something for each core?

too late already did it anyways- Just a way to organise what builds from source and what doesn't- just for getting started:

https://docs.google.com/spreadsheets/d/17f4qvN33DTjYeuCbbLfVyUEe4WUIKuwEpXKUykGfOBA/edit?usp=sharing

joolswills, 2015-12-08:
as Herb says - it is not yet production ready on Jessie. I have also never tested on the Jessie Lite, so it will need some more work im sure to work on there out of the box (due to some dependencies we miss - I will need to go through this). 

Not sure the spreadsheet will be of use to me - I prefer just to use the issue tracker etc. If you want to keep a spreadsheet though of course feel free.

If you post a debug log I can advise further (with build issues), but the issue with retroarch binaries is they need rebuilding as they were built with libjack, but we don't include a dependency for it. this is fixed in source though and the next binaries won't include this.

If you just want a working RetroPie, please use an image or start from Wheezy.

joolswills, 2015-12-08:
you can also install libjack-dev to fix the missing lib for retroarch.

joolswills, 2015-12-08:
RetroArch doesn't build from source currently on the rpi.

jimbo27, 2015-12-08:
@joolswills: after installing libjack-dev lr-snes9x-next (binary install) works! great! :+1:
lr-pocketsnes (binary install) still do not work for me. runcommand.log is empty...

Whatever: I will now start with a clean new Jessie Lite and try to install every component from source and binary. I will post my findings to Pastebin and those Pastbin links to the spreadsheet mentioned above. The spreadsheet will help to keep track of progress (thanks @HerbFargus )

joolswills, 2015-12-08:
Please can you wait until I have had a chance to go through it with Jessie lite - it will be easier for me to fix up anything obvious that way.

jimbo27, 2015-12-08:
Sure. No problem.

joolswills, 2016-01-04:
So the Jessie image is released. Everything has been tested - sorry I forgot to get back to you @jimbo27 - I went through everything and it looks like all the binaries are working ok. Feel free to do your own testing however.

I've not heard any Jessie specific bug reports yet, so I may well make it the official image for the next release so we can drop wheezy support (at least for prebuilt images / binaries)

jimbo27, 2016-01-05:
Test run 1: Installation of Emulationstation, RetroArch (and lr-snes9x-next for testing purpose) from source

Starting environment: Pi2, fresh installed Jessie Lite from 2015-11-21 (kernel 4.1), git and Retropie script installed + all necessary dependencies

Emulationstation: Build - OK (log: http://pastebin.com/q2E3fsnJ), Working - OK
RetroArch: Build - OK (log: http://pastebin.com/SefrQsfJ), Working - OK
lr-snes9x-next: Build - OK (log: http://pastebin.com/xqt8yUhK), Working - OK

Next test run: Installation of Emulationstation, RetroArch from binaries

jimbo27, 2016-01-05:
Test run 2: Installation of Emulationstation, RetroArch (and lr-snes9x-next for testing purpose) from binaries

Starting environment: Pi2, fresh installed Jessie Lite from 2015-11-21 (kernel 4.1), git and Retropie script installed + all necessary dependencies

Emulationstation: Build - OK (log: http://pastebin.com/WUv3hsdU), Working - NO (error while starting Emulationstation: "error while loading shared libraries: libSM.so.6: cannot open shared object file: No such file or directory")
RetroArch: Build - OK (log: http://pastebin.com/RSnu8zga), Working - ??
lr-snes9x-next: Build - OK (log: http://pastebin.com/BZQWhBBW), Working - ??

Next test run: Retest Installation of Emulationstation, RetroArch from binaries

joolswills, 2016-01-06:
no idea why it requires libsm - can't see anything in the source as a dependency on that but it must have decided to link with it. However installing libsm-dev would fix it. I don't think when installing from binary we are ever going to get the dependencies 100% if choosing just a couple of packages for reasons like this.

joolswills, 2016-01-06:
I will dig into this a bit more though - and see if I can find why it links with libSM and as a last resort add it as a dependency.

jimbo27, 2016-01-06:
Test run 3: Installation ofEmulationstation, RetroArch (and lr-snes9x-next for testing purpose) from binaries, 2nd try

Starting environment: Pi2, fresh installed Jessie Lite from 2015-11-21 (kernel 4.1), git and Retropie script installed + all necessary dependencies, apt-get update & upgrade, get latest of RetroPie script

Emulationstation: Build - OK (log: http://pastebin.com/uWH2zv3b), Working - NO (error while starting Emulationstation: "error while loading shared libraries: libSM.so.6: cannot open shared object file: No such file or directory")
RetroArch: Build - OK (log: http://pastebin.com/8Ev6tN3w), Working - ??
lr-snes9x-next: Build - OK (log: http://pastebin.com/HmKatxhM), Working - ??

Next test run: Install of libsm-dev

jimbo27, 2016-01-06:
Test run 4: Installation ofEmulationstation, RetroArch (and lr-snes9x-next for testing purpose) from binaries + install if libsm-dev

Starting environment: Result of test run 3

Emulationstation: Build - OK (log: http://pastebin.com/uWH2zv3b), Working - OK
RetroArch: Build - OK (log: http://pastebin.com/8Ev6tN3w), Working - OK
lr-snes9x-next: Build - OK (log: http://pastebin.com/HmKatxhM), Working - OK

Next test run: Installation of all Libretro cores from binaries

[19 more replies]

closed 2016-01-20 by joolswills