RetroPie/RetroPie-Setup #1032: *NOT AN ISSUE, SORRY I DIDNT KNOW WHERE ELSE TO CONTACT*

pancakeykakes, 2015-10-03:
I wanted to ask the contributors/devs of this project if they wanted to work with me on incorporating an idea using the script I wrote, but I don't quite think it qualifies for a pull request. We can talk here or via email.

Sorry again for posting it here. Still learning how github works and didn't see any other method of contact.

joolswills, 2015-10-03:
When you open an issue on github - it says "Please review the guidelines for contributing to this repository." - if you click the link it explains to use for forum for discussion

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

joolswills, 2015-10-03:
Better to start on the forum perhaps, and we can go from there. thanks.

pancakeykakes, 2015-10-03:
My apologies, I'm in mobile.

closed 2015-10-03 by pancakeykakes