RetroPie/RetroPie-Setup #1033: won't start raspi-config or configure wifi

EricxDu, 2015-10-04:
I've made a brand new RetroPie SD card and am trying to configure the install. First step is to go into "RetroPie" menu and start "Raspberry Pi Configuration Tool Raspi-Config", but it won't start. It drops into console and some messages said it can't install packages. In case it needed an internet connection, I stuck in my wi-fi dongle and selected "Configure Wifi", but that option failed with the same error messages.

I don't think RetroPie needs/should need an internet connection to configure wi-fi and/or use raspi-config. That'd be a bit odd.

joolswills, 2015-10-04:
It it shouldn't need any additional packages unless some have been removed. Which image did you download ?

joolswills, 2015-10-04:
I just tried with a fresh 3.0 image on a rpi2 without an internet connection, and was able to access raspi-config and wifi configuration from emulationstation without installing any additional packages.

EricxDu, 2015-10-04:
@joolswills Thanks for checking it out. I'm using an image I downloaded just this morning from RetroPie retropie-v3.0-rpi1.img

Only difference between us is I'm using Rpi1

HerbFargus, 2015-10-04:
I just set retropie 3.0 on my rpi1 just a couple days ago and had no issues- perhaps there was file corruption on download of your image.

EricxDu, 2015-10-04:
@HerbFargus Quite possible. It sounds like I have an isolated problem. I will check a couple of things now.

EricxDu, 2015-10-04:
Some specifics: I downloaded via the torrent link. I just verified local data and it seems to be fine.

I used Xarchiver to extract and dd to flash the image to an micro SD card.

The micro SD card is a Kingston 8GB which I'm fairly certain is on the Rasperry Pi compatibility list. I instert in into an adapter sleeve to fit into the Pi.

EricxDu, 2015-10-04:
I just re-flashed the card and tried the process again... this time there was no problem. How strange. Is there perhaps anything else to address before closing the issue?

HerbFargus, 2015-10-04:
glad you got it sorted. sometimes along the process it may just write improperly, might have been the case here.

joolswills, 2015-10-04:
glad you have it sorted.

closed 2015-10-04 by joolswills