RetroPie/RetroPie-Setup #1037: Can't proceed after Cofiguration-Screen (RP 2B)

qohelet, 2015-10-07:
Hi!
I've put the Image on the SD-Card, ran the setups (pi-config and the retropie-setup), installed the emulators (SNES) and copied some free roms I could find in the snes-folder. 

I tried several attempts just to start it (with keyboard, no gamepad) but till now could never proceed the white "Configure"-Screen. It's only possible to leave it entirely by pressing F4.

Do I need to press something on the keyboard? Or did I forget to change any settings? I also switched the boot-option in the retropie-setup, but nothing changed...

joolswills, 2015-10-07:
You don't need to install any emulators - they are all on the image.

what happens when you hold down a a key on the keyboard ? Have you tried plugging in a controller and holding down a button ?

qohelet, 2015-10-07:
I installed the emus via the setup.
Nothing happens. I can alter the controls via the keyboard but entering "OK" does not change anything.

Right now I'm not in the possession of a USB-Controller (some issues with the post) so I tried to use the plain keyboard (or wait another week).

joolswills, 2015-10-07:
If you download an image, all emulators are already present/installed.

So you have gone through configuring buttons on the keyboard ? At the end when you get to OK, have you pressed the button you configured as "A" ?

joolswills, 2015-10-07:
Did you follow this guide ? https://github.com/RetroPie/RetroPie-Setup/wiki/First-Installation

qohelet, 2015-10-07:
Aaaaah... got it, after finishing setting the controls I had to press the "A" button, not ANY button. Thx, problem solved...

closed 2015-10-07 by joolswills