RetroPie/RetroPie-Setup #1038: Xbox360 Controller sending input to terminal in background of ES

pancakeykakes, 2015-10-08:
Whenever I use my controller to navigate ES, retroarch menu, or play games, the controller's input is also being accepted by the terminal in the background.

For example, if I happen to press "up" on the D-pad and then "A", the terminal will recognize that as pressing "up" arrow and "enter" on the keyboard, effectively bringing up the last entered command and executing it.

As you can imagine, if my last command in bash was a reboot command, this can be quite problematic.

Anyone know of a way to stop this?

Here is what commands I am using for xboxdrv upon launching ES/RP:

xboxdrv --detach-kernel-driver --wid 0 --led 3 --trigger-as-button --ui-buttonmap RB=exec:/home/pi/emu.sh --ui-buttonmap LB=exec:/home/pi/kodi.sh --ui-buttonmap GUIDE=exec:/home/pi/killswitch.sh --quiet --silent & sleep 1

joolswills, 2015-10-08:
Please use the forum for support requests.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2015-10-08 by joolswills

pancakeykakes, 2015-10-08:
Soo the issue with my controller sending background input to the terminal only happens if Emulationstation is launched from a script.

If I manually launch by typing 'Emulationstation' in terminal, it doesn't happen.

pancakeykakes, 2015-10-08:
Even executing a script with the contents

#!/bin/bash
emulationstation

causes this issue.

pancakeykakes, 2015-10-08:
I would like to politely request we re-open this issue and investigate. It's not an xboxdrv bug.

joolswills, 2015-10-09:
I don't see how this is related to RetroPie-Setup - are you launching emulationstation in the background with & ? You are not using a default out of the box retropie-setup  but working with your own script/project correct?

pancakeykakes, 2015-10-09:
I installed RetroPie/es as a standalone environment on top of a fresh
Wheezy install.

As a way to make sure my little script wasn't causing conflicts, as well as
eliminate xboxdrv as a cause, I made a script that was literally just a
hashbang with the command "Emulationstation"

When running that way, All input, be it by controller or keyboard, is
extremely glitchy when opening RetroPie Setup, or launch options for ROMs.
The glitchy input is also seen being accepted by the terminal in
background, some of which actually bleeds through over the menus.

However, if I manually type "Emulationstation" as a command in terminal to
launch it, none of this happens.

Maybe Emulationstation needs to be launched with a more complex argument or
parameter when launching via script like that, but I think this is enough
to conclude that xbox drv isn't the cause.

If you would like, I can make a short video showing the behavior and post
it here/send it to you.

joolswills, 2015-10-09:
You have a problem with your own custom script - the keyboard input problem doesn't happen with a default retropie booting into emulationstation - so please use the forum as requested. thanks. I may be able to help, but this is not the right place.

pancakeykakes, 2015-10-09:
Alright, I posted here:
http://blog.petrockblock.com/forums/topic/input-bugs-when-launched-any-way-other-than-manual-bash-command/

I'm very sorry for being a pain in the ass about all of this, I am just extremely determined to figure this out, because I feel that I have a very useful project that many will enjoy, but I am too much of a perfectionist to let people have to deal with this issue should I release it.

Thanks for putting up with me.