RetroPie/RetroPie-Setup #1039: (R3.1) mupen64plus rice plugin missing

mauruco, 2015-10-09:
Hello, I was testing yesterday (Retropie 3.1) and I not find the rice plugin for mupen64plus. After some tests I installed rice plugin from the expirimental section. And now it works. But does not seem to be the same as before in 3.0.

joolswills, 2015-10-09:
using retropie 3.1 image ? rpi1 or rpi2 ?

mauruco, 2015-10-09:
Yes retropie 3.1 RPi2

joolswills, 2015-10-09:
Please can you detail your problem - what do you mean by you can't find the plugin ?

mauruco, 2015-10-09:
When starting a game change the emulator to mupen64plus-rice works. But the game does not run.

mauruco, 2015-10-09:
It seemed that the other two emulators (mupen64plus-rice and Mupen64-gles) are missing. They are in the selection menu, but do not exist.

joolswills, 2015-10-09:
there is only a single emulator - it runs it with different parameters depending on which is selected - ill check on a fresh 3.1 image

joolswills, 2015-10-09:
ok can confirm the problem - thanks - I saw this during development but thought I had resolved it - the problem is due to it linking against debian egl libraries if they are installed rather than the videocore ones I think. I will rebuild some working binaries.

joolswills, 2015-10-09:
ok. somewhat annoying - I had resolved this before release but forgot to upload the new binaries!

I have rebuilt again and uploaded now - if you reinstall the emulator from retropie-setup menu 5 from binary it should work again.

gizmo98, 2015-10-09:
What do you mean with "does not seem to be the same as before in 3.0"? The experimental rice plugin should be better.

mauruco, 2015-10-09:
Thank joolswills
gizmo 98, yes, now comparing with the previous (RP3.0) it's realy better. It was a feeling "hmm seems to work better".

closed 2015-10-09 by joolswills