RetroPie/RetroPie-Setup #1040: Add half life 1 engine

gizmo98, 2015-10-09:
Add half life1 engine for half life 1 engine based games:
https://github.com/ptitSeb/halflife

Nokturn, 2016-02-16:
Is it possible?

gizmo98, 2016-02-17:
I have not tested but there is a rpi2 Makefile.

Pygmaee, 2016-03-12:
Any body tested that? 
Would be awesome if Half-Life runs.

wetriner, 2016-09-04:
I spent some time looking into this, but didn't have much luck. Despite that, I'm fairly confident that this is possible to accomplish(but maybe impractical).

[This thread](https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=115767&sid=63157b4aeae587e9eb911a8b4d202682) shows the configuration process to get half-life to work on the pi based off of the working pandora port. The biggest problem with half-life seems to be the closed-source vgui.so that only runs on x86 architechture. Xash3d takes its place in making half-life possible on ARM.

Some things to note about the groundwork that was done. The setup utilized SDL compiled with OpenGL, and some components are dependent on Xserver. Unfortunately I have yet to see the framerate; my attempts always hung after a "fallback to dedicated server mode" message.

[This post](https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=115767&p=921751#p921751) in particular is the most up to date I could find for how to get it set up. Be aware, it did not work for me with the most up-to-date raspbian. `apt-get` had problems with the libjpeg8-dev package, and there's an extra zxvf on the `rm` line.

XD3l, 2016-09-04:
This should work in theory. I have Xash3D running on my Fire HD 6 and it
runs like a charm... all but with touch screen.
I'd love to see this, DOOM feels so lonesome without it. ;)

“Therefore do not deceive yourself! Of all deceivers fear most yourself!”
― Søren Kierkegaard
<https://www.goodreads.com/author/show/6172.S_ren_Kierkegaard>

AaronRays468, 2016-09-04:
Do you know? Q3A`s engine is modified into goldsrc

cesarpuig, 2019-11-21:
Hi, This work for me: https://github.com/sergiopoverony/xash3d
Resolution 800x600.