RetroPie/RetroPie-Setup #1042: Retropie audio

madmike008, 2015-10-11:
Hi, I'm sending this message to inform you that the 3.1 update fixed the snes game problem but now no audio will come out using the HDMI cable.

joolswills, 2015-10-11:
it's likely you need to change configuration such as adding  hdmi_drive=2 in /boot/config.txt

pancakeykakes, 2015-10-11:
If you still don't get sound in retroarch, you can force HDMI audio by
adding/modifying these lines to retroarch.cfg

This worked for me.

audio_rate_control = "true"
audio_rate_control_delta = "0.005000"
audio_max_timing_skew = "0.050000"
audio_volume = "0.000000"
audio_driver = "sdl"
audio_enable = "true"
audio_mute_enable = "false"
audio_out_rate = "44100"
audio_device = hw:0,1
hdmi_group=1
hdmi_mode=4

joolswills, 2015-10-11:
Please use the forum for further help as detailed in https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md - you will also find threads with information on fixing sound issues.

closed 2015-10-11 by joolswills