RetroPie/RetroPie-Setup #1043: unzip -L issue

satanasov, 2015-10-11:
There are few unzip scripts that use -L parameter for the command:

````
`--2015-10-11 13:31:08--  http://downloads.petrockblock.com/retropiearchives/zork3.zip
Resolving downloads.petrockblock.com (downloads.petrockblock.com)... 46.252.21.198
Connecting to downloads.petrockblock.com (downloads.petrockblock.com)|46.252.21.198|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 77155 (75K) [application/zip]
Saving to: ‘zork3.zip’

     0K .......... .......... .......... .......... .......... 66% 99.3K 0s
    50K .......... .......... .....                           100%  538K=0.6s

2015-10-11 13:31:08 (137 KB/s) - ‘zork3.zip’ saved [77155/77155]

unzip: invalid option -- 'L'
BusyBox v1.22.0.git (2013-12-25 06:52:05 CET) multi-call binary.

Usage: unzip [-lnopq] FILE[.zip] [FILE]... [-x FILE...] [-d DIR]

Extract FILEs from ZIP archive

        -l      List contents (with -q for short form)
        -n      Never overwrite files (default: ask)
        -o      Overwrite
        -p      Print to stdout
        -q      Quiet
        -x FILE Exclude FILEs
        -d DIR  Extract into DIR

mv: cannot stat ‘data/zork3.dat’: No such file or directory

````

Proposition - change them to -l

joolswills, 2015-10-11:
Why are you using busybox - what set up is this ? retropie-setup is designed for running with bash - and so not designed for the busybox built in unzip.

satanasov, 2015-10-11:
Raspberry Pi 2
OS: Xbian (current).

joolswills, 2015-10-11:
RetroPie-Setup is not designs for running on Xbian. It runs on Raspbian Wheezy.

satanasov, 2015-10-11:
OK ... but this is the only issue I have faced with RetroPie-Setup ... it appears they are almost compatible :)

Thanks anyway and sorry for the bother.

joolswills, 2015-10-11:
You could try using bash instead of busybox. we use unzip -L to lowercase on extraction - It is likely other stuff will also not work correctly without the use of bash.

closed 2015-10-11 by joolswills