RetroPie/RetroPie-Setup #1048: amstrad emulation failing

cheekymonkat, 2015-10-15:
Hi, I performed an upgrade and also a new SD image install - but it appears that the amstrad emulation now just fails and returns to ES screen.  happens for all the games.

I've the latest PI.

Anyone else seen this?

Also - is there a download available for SD 3.0 as I'd like to go back.

Thanks

joolswills, 2015-10-15:
anything in /tmp/runcommand.log ?

cheekymonkat, 2015-10-15:
yes (I decided to look after sending the email!)

/opt/retropie/emulators/cpc/cpc4rpi: error while loading shared libraries:
libEGL.so.1: cannot open shared object file: No such file or directory

Thanks Alan

joolswills, 2015-10-15:
I will have a new binary up in a minute. I know what the problem is.

cheekymonkat, 2015-10-15:
awesome thanks!

joolswills, 2015-10-15:
Please go to retropie-setup, menu 5, and choose cpc then install from binary.

joolswills, 2015-10-15:
BTW it would have fixed it had you rebuilt it from source - it failed due to it linking with the wrong EGL library on my dev system. However it doesn't even need to link to EGL, so I removed the dependency. This emulator will be replaced at some point with Caprice32 (it is basically a renamed caprice32). There is also a libretro Caprice32 emulator you can try also with retropie (switch from pre-launch menu)

cheekymonkat, 2015-10-15:
great thanks! worked a treat.  Thanks for your rapid response.  Great tool!

cheekymonkat, 2015-10-15:
Sorry - have a new issue (maybe related) - I get no sound at all?  I hear it from other emulators - any ideas?

Thanks

joolswills, 2015-10-15:
Via HDMI or headphone ? Did you set the audio configuration correctly  (From retropie-setup / configuration)

joolswills, 2015-10-15:
There is sound on it, I can confirm here, so it sounds like a configuration issue your end

cheekymonkat, 2015-10-15:
Ah, yes - sorry - I was getting sounds for other emulators from the HDMI connection but once I switched to headphone it worked... sorry my mistake!
Thanks

closed 2015-10-15 by cheekymonkat