RetroPie/RetroPie-Setup #1050: Add OpenTyrian to Ports

HerbFargus, 2015-10-17:
I am taking my info from here: http://dosonthepi.blogspot.com/2015/02/install-opentyrian.html

### Install files
```
git clone https://github.com/vanfanel/SDL12-kms-dispmanx.git
sudo mv ~/SDL12-kms-dispmanx/ /opt/sdl-dispmanx
cd /opt/sdl-dispmanx
CFLAGS="-Ofast -mcpu=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard" ./configure --prefix=/opt/sdl-dispmanx --disable-video-opengl --enable-video-dispmanx --disable-video-fbcon --disable-video-kms --disable-video-directfb --disable-oss --disable-alsatest --disable-pulseaudio --disable-pulseaudio-shared --disable-arts --disable-nas --disable-esd --disable-nas-shared --disable-diskaudio --disable-dummyaudio --disable-mintaudio --disable-video-x11 --disable-input-tslib
make && sudo make install
cd
wget -c https://sites.google.com/site/dosonthepi/opentyrian.zip
unzip opentyrian.zip
cd ~/opentyrian
make release
wget -c http://www.camanis.net/tyrian/tyrian21.zip
unzip -j tyrian21.zip -d data
```

### Configure for Retropie
```
cd ~/RetroPie/roms/ports
sudo nano opentyrian.sh
#!/bin/bash
cd /home/pi/opentyrian
/opt/retropie/supplementary/runcommand/runcommand.sh 0 "./opentyrian"
sudo chmod +x opentyrian.sh
```

I'm just posting this as a basis for a module- placing it in a module should simplify the code a bit.

HerbFargus, 2015-10-17:
This is my preliminary attempt at a module: it needs a bit of work:

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="opentyrian"
rp_module_desc="Tyrian"
rp_module_menus="4+"

function depends_opentyrian() {
}

function sources_opentyrian() {
    gitPullOrClone "$md_build" https://github.com/vanfanel/SDL12-kms-dispmanx.git
    cd "$md_build"
    CFLAGS="-Ofast -mcpu=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard" ./configure --prefix=/opt/sdl-dispmanx --disable-video-opengl --enable-video-dispmanx --disable-video-fbcon --disable-video-kms --disable-video-directfb --disable-oss --disable-alsatest --disable-pulseaudio --disable-pulseaudio-shared --disable-arts --disable-nas --disable-esd --disable-nas-shared --disable-diskaudio --disable-dummyaudio --disable-mintaudio --disable-video-x11 --disable-input-tslib
    make
    make install
}

function build_opentyrian() {
}

function install_opentyrian() {
    wget -c https://sites.google.com/site/dosonthepi/opentyrian.zip
    unzip -o opentyrian.zip -d "$md_inst"
    rm opentyrian.zip
    make release

    wget -c http://www.camanis.net/tyrian/tyrian21.zip
    unzip -j tyrian21.zip -d data
    rm tyrian21.zip
}

function configure_opentyrian() {

    mkRomDir "opentyrian"
    mkRomDir "ports"

    cat > "$romdir/ports/OpenTyrian.sh" << _EOF_
#!/bin/bash
/opt/retropie/supplementary/runcommand/runcommand.sh 0 "./opentyrian"
_EOF_

    chmod +x "$romdir/ports/OpenTyrian.sh"

    setESSystem 'Ports' 'ports' '~/RetroPie/roms/ports' '.sh .SH' '%ROM%' 'pc' 'ports'    
}
```

dudleydes, 2015-10-17:
I have revisited OpenTyrian over the last week and can make the following observations.

I have not been able to compile a binary on a RetroPie 3.x image using the latest source available from https://bitbucket.org/opentyrian/opentyrian/downloads. I am getting an internal compiler error. You can find the console output [here](http://pastebin.com/tx0F2Wax).

I have returned to the source code I downloaded in February 2015 and have made some changes to the code that builds a binary that supports resolutions up to 1280 x 800. I have written and tested a scriptmodule (below). Here is a list of changes I made.

1. I have amended the Makefile (line 19) so that it points to the SDL1.2.15 library with rpi fixes and dispmanx (/usr/bin/sdl-config) that comes installed with RetroPie images rather than building vanfanel's library.

2. Hqx scaling filters are disabled by commenting out lines 53, 56 & 58 in video_scale.c. I have left the other scalers intact so that OpenTyrian can be played in the following resolutions:
  * Scaler: None >  Resolution:  320 x 208
  * Scaler: 2x >  Resolution:  640 x 400
  * Scaler: 3x >  Resolution:  960 x 608
  * Scaler: 4x >  Resolution:  1280 x 800

   Note that the resolutions for scalers None and 3x are non-standard because dispmanx requires the Y axis to be a multiple of 16.

   Smoothing scalers are also available for 640 x 400 (Scale2x) and 960 x 608 (Scale3x).

3. Dispmanx flag needs to be added and enabled by default otherwise OpenTyrian will not launch. There will be a "failed to initialize any supported video mode" error.

4. The OpenTyrian binary will be located in the /opt/retropie/ports/opentyrian folder. The binary is installed with the cp command (I couldn't work out how to create make install rules).

5. Freeware game data is downloaded to /home/pi/RetroPie/roms/ports folder. The shell script to launch OpenTyrian has a --data switch to make sure the binary finds the game data.

Here is the scriptmodule.

```
#!/usr/bin/env bash

# This file is part of RetroPie.
# 
# (c) Copyright 2012-2015  Florian Müller (contact@petrockblock.com)
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/petrockblog/RetroPie-Setup/master/LICENSE.md.
#

rp_module_id="opentyrian"
rp_module_desc="Open Tyrian, port of the DOS shoot-em-up Tyrian"
rp_module_menus="4+"
rp_module_flags="dispmanx"

function depends_opentyrian() {
    getDepends libsdl1.2-dev
}

function sources_opentyrian() {
    wget -O- -q "https://sites.google.com/site/dosonthepi/opentyrian.tar.gz" | tar -xvz --strip-components=1
}

function build_opentyrian() {
    make clean
    make
    md_ret_require="$md_build/opentyrian"
}

function install_opentyrian() {
    cp opentyrian $md_inst/opentyrian
}

function configure_opentyrian() {
    mkRomDir "ports/opentyrian"

    # Get Tyrian 2.1 (freeware game data)
    pushd $romdir/ports/opentyrian/
    wget -c http://www.camanis.net/tyrian/tyrian21.zip
    unzip -j tyrian21.zip -d data
    rm tyrian21.zip

    cat > "$romdir/ports/OpenTyrian.sh" << _EOF_
#!/bin/bash
$rootdir/supplementary/runcommand/runcommand.sh 0 "$md_inst/opentyrian --data $romdir/ports/opentyrian/data"
_EOF_

    # Set startup script permissions
    chmod +x "$romdir/ports/OpenTyrian.sh"
    chown $user:$user "$romdir/ports/OpenTyrian.sh"

    # Enable dispmanx by default.
    setDispmanx "$md_id" 1

    # Add OpenTyrian to EmulationStation
    setESSystem 'Ports' 'ports' '~/RetroPie/roms/ports' '.sh .SH' '%ROM%' 'pc' 'ports'
}
```

joolswills, 2015-10-17:
Probably can patch it up to work without the dispmanx backend - might need a couple of SDL changes to the screen/window opening code.

To avoid the compiler fail, try using gcc 4.8 or newer

joolswills, 2015-10-17:
adding a new framebuffer mode to match 320 x 208 may also help running without dispmanx btw.

dudleydes, 2015-10-18:
I have managed to build OpenTyrian from the latest source. I had to remove all references to hqNx in the source code, namely removing the file video_scale_hqNx.c and commenting out lines 53, 56 and 58 in video_scale.c (otherwise I get an undefined reference error while compiling).

The sources section of the script module would now be:

```
function sources_opentyrian() {
    wget "https://bitbucket.org/opentyrian/opentyrian/get/98afb31c7343.zip"
    unzip 98afb31c7343.zip
    rm 98afb31c7343.zip
    pushd opentyrian-opentyrian-98afb31c7343/src
    # Remove references to hqNx
    rm video_scale_hqNx.c
    # Comment out lines 53, 56 and 58 in video_scale.c
    popd
}
```
Code needs to added to comment out the lines in video_scale.c - I've not been able to work out how to do this.

The install section will still need work. I've not been able to make progress here as I can't make my way through the makefile.

joolswills, 2015-10-20:
Thanks. Will look at it when I get some free time.

joolswills, 2015-10-20:
builds fine here without changes - you are probably running out of free memory or something so that file isn't compiling.

joolswills, 2015-10-20:
(I'm sorting out a module now btw).

joolswills, 2015-10-20:
https://github.com/RetroPie/RetroPie-Setup/pull/1056

joolswills, 2015-10-21:
Now merged. thanks.

joolswills, 2015-10-21:
continuing on here from the pull request -

if you update retropie-setup I have added swap if memory is low to build now. if it still runs out it will need increasing further

dudleydes, 2015-10-21:
OpenTyrian now builds on Rpi1.

I'd like to mention that the build time was one hour 4 minutes and using the hq2x scaler makes the game unplayable. Without the hqNx elements, the build time is 7 minutes.

joolswills, 2015-10-21:
yeh it might be worth removing it still. gcc optimiser struggles with it

HerbFargus, 2015-10-31:
Would it be worth forking and removing those bits of code or is it something that we would just need to add in as a patch as it compiles?

joolswills, 2015-10-31:
one of the above, or neither :)

joolswills, 2015-10-31:
It only takes 3 mins to build on a rpi2, and someone might want to use say hq2x so it looks less blurry than using no scaler. Not sure if I'm currently too worried about a longer built time on a rpi1.

we could maybe disable it just for the rpi1, but it's not something I consider a priority right now

HerbFargus, 2015-10-31:
fair enough

joolswills, 2015-10-31:
for the rpi1, people probably want to switch from scale2x (default), to none also for best performance.

joolswills, 2015-11-07:
ill close this for now. ill revisit the compile time at some other point.

closed 2015-11-07 by joolswills