RetroPie/RetroPie-Setup #1053: SNESDev-RPi not installling

smartroad, 2015-10-18:
Hi,

Just put this onto a new install of Wheezy and trying to install SNESDev but it returned with an error:

Could not successfully configure SNESDev (Driver for the RetroPie GPIO-Adaptor)
(/home/pi/RetroPie-Setup/tmp/build/snesdev/src/SNESDev not found)

Have been able to manually install it following the instructions on its git entry though

joolswills, 2015-10-18:
Need the full log (stored in RetroPie-Setup/logs) post it to pastebin.com or similar.

joolswills, 2015-10-18:
I have reproduced the problem - seems it fails on the first built, but if you try it again it works. This is due to library dependencies and build order. I will feed it upstream

joolswills, 2015-10-18:
Aha. it was due to retropie-setup attempting to build in parallel. This should be fixed if you update retropie-setup now.

closed 2015-10-20 by joolswills

smartroad, 2015-10-23:
Sorry only just had time to try this. It hasn't worked for me (new install from 3.1 image on Pi2). Updated script and had:

Log started at: Fri Oct 23 14:39:29 UTC 2015

= = = = = = = = = = = = = = = = = = = = =
Configuring 'snesdev' : SNESDev (Driver for the RetroPie GPIO-Adapter)
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/petrockblog/SNESDev-RPi.git" "/home/pi/RetroPie-Setup/tmp/build/snesdev"
Cloning into '/home/pi/RetroPie-Setup/tmp/build/snesdev'...
make: *** No targets specified and no makefile found.  Stop.
cp: cannot stat `src/SNESDev': No such file or directory
cp: cannot stat `src/Makefile': No such file or directory
cp: cannot stat `Makefile': No such file or directory
cp: cannot stat `scripts/Makefile': No such file or directory
cp: cannot stat `scripts/SNESDev': No such file or directory
cp: cannot stat `supplementary/snesdev.cfg': No such file or directory
/opt/retropie/supplementary/snesdev /home/pi/RetroPie-Setup
make: *** No rule to make target 'install'.  Stop.
/home/pi/RetroPie-Setup
make: *** No rule to make target 'uninstallservice'.  Stop.
Could not successfully configure SNESDev (Driver for the RetroPie GPIO-Adapter) (/home/pi/RetroPie-Setup/tmp/build/snesdev/src/SNESDev not found).

Log ended at: Fri Oct 23 14:39:44 UTC 2015
Total running time: 0 hours, 0 mins, 15 secs

in the log

joolswills, 2015-10-23:
I can only think you had a networking issue as it looks as though the git pull failed.

joolswills, 2015-10-23:
no my bad, there is an issue still when launching it from the setup menu. Will sort.

smartroad, 2015-10-23:
Ok :grin: was about to mention I was able to pull down the git directly so not an internet issue :wink:

joolswills, 2015-10-23:
ok should be sorted now.

smartroad, 2015-10-23:
Doesn't seem to be trying to install it?

Tried to enable on boot (only pads) and got the log:

```
Log started at: Fri Oct 23 15:05:09 UTC 2015
/opt/retropie/supplementary/snesdev /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Configuring 'snesdev' : SNESDev (Driver for the RetroPie GPIO-Adapter)
= = = = = = = = = = = = = = = = = = = = =

[H[J
= = = = = = = = = = = = = = = = = = = = =
Enabling SNESDev on boot.
= = = = = = = = = = = = = = = = = = = = =

Setting button_enabled=0 in /etc/snesdev.cfg
Setting gamepad1_enabled=1 in /etc/snesdev.cfg
Setting gamepad2_enabled=1 in /etc/snesdev.cfg
make: *** No rule to make target 'installservice'.  Stop.
/home/pi/RetroPie-Setup

Log ended at: Fri Oct 23 15:05:16 UTC 2015
Total running time: 0 hours, 0 mins, 7 secs
```

Is there an option to install SNESDev? It just occurred to me that all the options only say enable/disable not install in the SNESDev menu option

joolswills, 2015-10-23:
rm -rf /opt/retropie/supplementary/snesdev - probably due to left over stuff from previous fail. I'll make the check stricter.

joolswills, 2015-10-23:
it installs by default, but the check isn't great. I've just improved that. This module hasn't seen as much attention as some of the other stuff as I don't use the driver - and we used to ship with a binary of it by default which has changed and has shown up a few bugs.

smartroad, 2015-10-23:
That got it sorted! Many thanks :grin:

joolswills, 2015-10-23:
thanks for report.