RetroPie/RetroPie-Setup #1059: Menus on gamelist scrolling time to time in automatically way

LodanZark, 2015-10-24:
The Menus on gamelist scrolling time to time in automatically way, without my intervention...
Is it a feature or a bug?
If is a feature how can i disable it?

joolswills, 2015-10-24:
You have a buggy/broken controller.

LodanZark, 2015-10-24:
I am using a SNES Buffalo usb controller, I just tested on windows over joy.cpl and its working properly, I even touched the cables meanwhile plug in to see if would trigger some weird behaviors and non issue was found, anyway I will try a different gamepad just to make sure.

HerbFargus, 2015-10-24:
I had some old buggy controllers that did the same thing, I switched controllers and have had no issues since. The other thing you can try is reconfiguring your controller but not sure if that will help at all if it is a hardware issue.

LodanZark, 2015-10-24:
I have 2 Buffalo USB controllers, one does the trigger that issue and other one don't, the odd thing is that issue isn't affected in gameplay since I left an RPG opened in afk mode and the character didn't move or trigger any action, plus both gamepads works very good in gamepad callibration software on windows xD go figure...

Tele42, 2015-10-24:
Hypothetically, the windows driver could have implemented a quirks-mode and automatically detects the issue, while on linux, either a generic kernel module is used or a best-effort was made with the hardware whoever made the module had at the time. I do not have controllers like this, so I can only speculate.

joolswills, 2015-10-30:
nothing we can do about it currently - if it is a known hardware issue that can be fixed in the driver, this needs to be fed upstream to the linux kernel etc.

closed 2015-10-30 by joolswills