RetroPie/RetroPie-Setup #1060: Retroarch: Use analog stick as a dpad (if system supports no analog sticks)

gizmo98, 2015-10-24:
The retroarch option `input_player1_analog_dpad_mode = "1"`tells retroarch to use the left analog stick  as a dpad. (0=no remapping, 1=left analog stick, 2=right analog stick). The actual dpad works at the same time. This option does not interfere if a joypad without analog stick is used. For up to eight joypads it is only necessary to add the following lines:

configs/all/retroarch
```
input_player1_analog_dpad_mode = "1"
input_player2_analog_dpad_mode = "1"
input_player3_analog_dpad_mode = "1"
input_player4_analog_dpad_mode = "1"
input_player5_analog_dpad_mode = "1"
input_player6_analog_dpad_mode = "1"
input_player7_analog_dpad_mode = "1"
input_player8_analog_dpad_mode = "1"  
```

There are only two systems which use analog sticks. The option should be disabled in there retroarch.cfgs:
configs/psx/retroarch and configs/n64/retroarch
```
input_player1_analog_dpad_mode = "0"
input_player2_analog_dpad_mode = "0"
input_player3_analog_dpad_mode = "0"
input_player4_analog_dpad_mode = "0"
input_player5_analog_dpad_mode = "0"
input_player6_analog_dpad_mode = "0"
input_player7_analog_dpad_mode = "0"
input_player8_analog_dpad_mode = "0"  
```

HerbFargus, 2015-10-24:
+1 from me

gizmo98, 2015-10-25:
Verified lr-mupen64plus and lr-psx-rearmed ignore this option. So only configs/all/retroatch.cfg must be modified. If nothing speaks against i would like to add something like this:
```
for i in {1..8}
do
    iniSet "input_player$i_analog_dpad_mode" "1"
done 
```

joolswills, 2015-10-28:
This has broken D-pads on snes style controllers. What I am confused about is that you seem to have known about the issue beforehand ? https://github.com/libretro/RetroArch/issues/609#issuecomment-37754495

joolswills, 2015-10-28:
Reverted for now in https://github.com/RetroPie/RetroPie-Setup/commit/897934b0dbf913e01b659567e1928d0f4a82b6c5 - building a 3.2.1 image - will make some manual fix instructions for anyone that downloaded 3.2

gizmo98, 2015-10-28:
Will take a look into retroarch code. Makes no sense if it is working if normal dpad is bound to buttons but does not work if dpad is bound to axis (as i remember snes controllee use axis)

joolswills, 2015-10-28:
retroarch doesn't build for rpi currently btw but I have a pull request in with a fix.

gizmo98, 2015-10-28:
Oops. There is already an open issue:
https://github.com/libretro/RetroArch/issues/1780

gizmo98, 2015-10-28:
And the right position to fix this should be here:
https://github.com/libretro/RetroArch/blob/e84b860bf61f6bc276a344d2defd8b1ce9f17ef2/input/input_overlay.c#L1552

Fix could be "If there is no analog stick set input.analog_dpad_mode to 0(default)" or something equal.

gizmo98, 2015-10-28:
Or it happens here:https://github.com/libretro/RetroArch/blob/e84b860bf61f6bc276a344d2defd8b1ce9f17ef2/input/input_common.c#L473

joyaxis will be overwritten. In any case there should be a check if there are analog sticks.

gizmo98, 2015-10-28:
I have a test fix. Still runs with my controllers. Needs to be tested with a SNES clone.
https://github.com/gizmo98/RetroArch/commit/98d9d89d0db20050e0266438adbf42e18d93ce3f

Expectation:
If autoconfig file has no values for `input_l_x_plus_axis, input_l_x_minus_axis, input_l_y_plus_axis, input_l_y_minus_axis` key bindings of this four directions are equal. If `binds[RARCH_ANALOG_#_#_#].joyaxis` are equal don't overwrite key bindings of `binds[RETRO_DEVICE_ID_JOYPAD_UP].joyaxis, binds[RETRO_DEVICE_ID_JOYPAD_DOWN].joyaxis, binds[RETRO_DEVICE_ID_JOYPAD_LEFT].joyaxis, binds[RETRO_DEVICE_ID_JOYPAD_RIGHT].joyaxis`.

joolswills, 2015-10-28:
Sounds good - Thanks - if accepted upstream (don't want to maintain lots of retroarch patches etc) :)

gizmo98, 2015-10-28:
I cannot test until my new controller arrives. :-)
It don't harms my current controllers but i don't know if it is fixes snes clone bug.

HerbFargus, 2015-10-28:
I've got a cheap snes usb controller as well as a few logitech f310 controllers. I may be able to give it a test when I find a free moment

HerbFargus, 2015-10-29:
Didn't look like it worked. dpad still doesnt work with the changes you made to retroarch. 

I tested with donkey kong with my snes controller on the snes-9x-next emulator, But my dpad and analog both work on my logitech f310.

HerbFargus, 2015-10-29:
Just for reference this is my autoconfig file for my snes:

```
input_device = "USB gamepad           "
input_driver = "udev"
input_r_btn = "5"
input_save_state_btn = "5"
input_start_btn = "9"
input_exit_emulator_btn = "9"
input_l_btn = "4"
input_load_state_btn = "4"
input_up_axis = "-1"
input_a_btn = "1"
input_b_btn = "2"
input_reset_btn = "2"
input_down_axis = "+1"
input_right_axis = "+0"
input_state_slot_increase_axis = "+0"
input_x_btn = "0"
input_menu_toggle_btn = "0"
input_select_btn = "8"
input_enable_hotkey_btn = "8"
input_y_btn = "3"
input_left_axis = "-0"
input_state_slot_decrease_axis = "-0"
```

gizmo98, 2015-10-29:
Thank you herb! A snes controller arrived today. I modified code a little bit. Should run now. Tested with a ps3 controller and a snes clone.

HerbFargus, 2015-10-29:
I'll give it a test tonight when I get off work- always good to have a second opinion ;)

HerbFargus, 2015-10-30:
Brilliant Mate :) I can confirm this works with only my snes controller plugged in, only my f310 controller plugged in, and both plugged in. 

Also as an aside I can confirm the search box fix works as well.

joolswills, 2015-10-30:
I see the change has gone upstream. I will build a new binary for jessie, and then we can put the setting back. One more thing, does everything work as it should with emulators such as mame where you might want to use the analogue stick as a trackball etc? This needs to be checked also.

gizmo98, 2015-10-30:
After my current code audit i would recommend to disable this setting for mame and lr-mupen64plus.

joolswills, 2015-10-30:
rather than setting it by default, and then disabling for some cores, it might be better we just add it to the config editor, so it can easily be enabled/disabled as needed ? Just want to avoid any problems we might not have come across yet or other emulators which it might affect.

joolswills, 2015-10-30:
btw with your success in modifying the input code in retroarch - how about this one https://github.com/libretro/RetroArch/issues/2230 ;-) Not had a chance to look yet - but the last post is interesting - perhaps you know how to do that https://github.com/libretro/RetroArch/issues/2230#issuecomment-149635177

gizmo98, 2015-10-30:
It should be always disabled if a system has independent analog sticks and dpads like psp, n64, dreamcast, xbox,  and mame(?). 
Retroarch maps DPAD to RETRO_DEVICE_ID_JOYPAD, "analog left" to RARCH_ANALOG_LEFT and "analog right" to RARCH_ANALOG_RIGHT. If you enable "analog_dpad_mode=1" retroarch maps "analog left" to RARCH_ANALOG_LEFT ""AND"" RETRO_DEVICE_ID_JOYPAD. So "analog left" is mapped to two different functions. There is no problem if emulated system does not support analog sticks. If you have a system which has dpads and analog sticks a dpad and analog stick event will be triggered if you use "analog stick left". 
I have not seen any problem with lr-mupen64plus because most n64 games don't use the dpad. I think funny things will happen if a game uses dpad and analog stick for different use cases.

So in my opinion analog_dpad_mode can be enabled for all systems except lr-mupen64plus, lr-ppsspp and lr-mame. We should test the new retroarch code for some days and decide then.

gizmo98, 2015-10-30:
And you need a multiple key press check for problem #2230? Seems possible. :-D

joolswills, 2015-10-30:
If it's added to the config editor, it's easily enabled for those that want it, and they can decide on which cores it;s enabled. It avoids problems - there will always be unknowns, and should a core come out that behaviour changes or is broken, we wont have to mess around disabling this on a per core basis which would be a pain since we don't overwrite users configs etc.

gizmo98, 2015-10-30:
I see. Have not played around with config editor until now. It's a good idea to add option there.

joolswills, 2015-10-31:
I have added it to the configuration editor - so I'm happy to leave it as that.

closed 2015-10-31 by gizmo98