RetroPie/RetroPie-Setup #1061: mupen64plus: add hotkey combos for exit, save state and load state

gizmo98, 2015-10-25:
Even if mupen64plus-input-sdl knows your controller there are no hotkeys for exit, save and load at the moment. The mupen64plus-core is responsible for exit, load and save hotkeys and it is working a little bit different. The setup of a exit hotkey for two ps3 controllers looks like "J0B16,J1B16". If you have controllers with different layouts it is important which one was connected first because the joypad number changes. If you connect in the wrong order buttons are wrong --> "J0B16,J1B1" or "J0B1,J1B16". Do overcome the "connect in the right order" problem i have written a udev rule which runs if a controller is added or removed. It uses controller config of emulationstation and writes known "select+start=exit", "select+L=load state" and "select+R=save state" combos to mupen64plus.cfg.     

I have opened up a repo for testing:
```
git clone --depth=1 https://github.com/gizmo98/mupen64plus-retropie-hotkeys.git
cd mupen64plus-retropie-hotkeys
sudo ./install.sh
```

If it does not work or you dislike this behavior type:
```
cd mupen64plus-retropie-hotkeys
sudo ./remove.sh 
```
Any suggestions?

HerbFargus, 2015-10-27:
I guess I'm not familiar enough with the configs in order to get this working. Do I need to reinstall mupen64 from source or do I have to set manual configs in mupen64plus.cfg?

My controller is logitech f310 so it is in the inputautocfg,ini so do I still need to set controls manually to incorporate hotkey support? Is what you discussed above related to this commit? https://github.com/mupen64plus/mupen64plus-core/commit/14bf40b236af1d5a5e7b86cd15a6cb628b4153fd

gizmo98, 2015-10-27:
You have found the right mupen64plus commit! Which video plugin are you using? I had no crash with GLideN64. 

A reboot is not really necessary. Just disconnect and reconnect your controller. udev rule will be triggered if a joypad/joystick device is added or removed. 

Your config file looks good!

HerbFargus, 2015-10-27:
I was on gles2n64 playing mario kart

gizmo98, 2015-11-22:
I rewrote the script. It's a normal bash script with hat and axis supports now.

HerbFargus, 2015-11-22:
Brilliant :+1:  Even works with the second player too. still crashes on load state but I think thats a mupen64plus bug unrelated to the hotkeys. (happened with glesn64 as well as gliden64)

gizmo98, 2015-11-22:
A crash should be core related. I had no crash but i still have a rp 3.1 image with updated modules. I will update mupen64plus to check if there is a regression.

gizmo98, 2015-11-23:
@joolswills. Do you think this script can be added?

It does the following things:
1. Note down every /dev/input/js* device
2. Note down device name of every found js device
3. Try to find a retroarch auto config file which contains device name
4. If file exists use iniGet to extract enable_hotkey exit_hotkey load_state_hotkey save_state_hotkey 
5. Convert to mupen64plus-core syntax
6. Write converted hotkeys to mupen64plus.cfg with iniSet

Script runs if a controller is added or removed (udev rule) and at startup (.bashrc).

joolswills, 2015-11-23:
if there is a udev rule why does it need to run at startup too ? (.bashrc would be the wrong place as that is bash specific).

I checked the script and I think it needs some work before it is included. Having yet another copy of the ini functions isn't ideal, so I would want to come up with a way to avoid that (and also for existing scripts to change it so it's not duplicated which has been on my mental todo list). Also littering with more udev hooks etc just for a single emulator but to run on joystick hotplugging. Why not just process it before the emulator is launched for example.

joolswills, 2015-11-23:
My feeling is also, as this is basically a problem in the way mupen64plus handles the hotkeys it would probably be a good idea to make it so the hotkeys can be set in the InputAutoCfg.ini on a per controller basis in the same way retroarch does - have you had any discussion with upstream ?

gizmo98, 2015-11-24:
Let me explain this. Emus use two kinds of controller setups:
-static: Just use device found under /dev/input (js0-js*). You don't know the name of the device. Setup must be reconfigured if you add/remove new controllers of equal or different type.
(gngeo, mupen64plus-core, gpsp, etc...)
-dynamic: Use device name of device. If device name matches a auto config use this config for device (js0-js*).
(retroarch, mupen64plus-input-sdl, emulationstation, reicast)

This script adds dynamic setup for mupen64plus-core. 

InputAutoConfig cannot be used because hotkey and input handling are done in different plugins (different plugin, different author?). mupen64plus-core only supports static setup and does not use InputAutoConfig.ini:
```
mupen64plus.cfg
Joy Mapping Stop = "J0B7/B6"
```
```
InputAutoCfg.ini
[Logitech Gamepad F310]
Start = button(7)
```

Running before emulator starts would be a good idea to remove udev rule and .bashrc entry!

joolswills, 2015-11-24:
I realise it's a different core and understand why you made it, but I think we have a somewhat ove engineered solution to a problem that should be fixed elsewhere first (or at least visited). As they do use dynamic joysticks in their input plugin it would make sense to at least visit that route. both code repos are on the same repository and have authors in common too.

This doesn't mean I don't appreciate all efforts :)

joolswills, 2015-11-24:
Anyway, if you submit it as a pull request without the udev bits, I'll accept it (after a code tweaks), at least until we have something better.

gizmo98, 2015-11-24:
I will open a feature request and a PR. I think we can use this script until mupen64plus team adds a better solution.

joolswills, 2015-11-24:
can include it in retropie-setup as part of the mupen64plus emu (subfolder of emulators perhaps like with the scripts in supplementary)

closed 2015-11-25 by gizmo98

joolswills, 2016-07-29:
@gizmo98 I have a related question - currently the "Joy Mapping Stop" is generated on Emulator launch based on the RetroArch config - however for 8bitdo devices that require the "hack" in the retroarch config - it will mean the wrong values are generated. I was thinking to either subtract 11, or perhaps we should generate this config based on the mupen64plus autogenerated config instead of retroarch ?

joolswills, 2016-07-29:
Looks like getBind could do with de-duplicating too - an additional outer loop for each hotkey and some changes should avoid the similar case statements.