RetroPie/RetroPie-Setup #1066: Libretro PPSSPP - Failed to load ui_atlas.zim

robertybob, 2015-10-28:
When I attempt to load a PSP game through the Libretro PPSSPP I get an error message:

Failed to load ui_atlas.zim

and then a black screen. Has anyone else had this message, or did my build fail when it was compiling?

http://libretro.com/forums/showthread.php?t=1960
https://www.reddit.com/r/emulation/comments/2wtqgd/ppsspp_retroarch_core_problems/

joolswills, 2015-10-29:
This was fixed as of https://github.com/RetroPie/RetroPie-Setup/commit/4b8a1d984e1671daa47dabce22600821bd823fdc

I guess you built before that commit ? You might want to rebuild as some other stuff has changed to - updating retropie-setup first.

PPSSPP  runs better than the libretro core btw - but the libretro core definitely works - try "Zarius burst"

robertybob, 2015-10-30:
Yeah you're right, just rebuilt it and seems to be ok. Was running Little Big Planet and was impressed by the speed (some occasional but very brief slowdowns).

Who's the person to ask to update the Wiki regarding PPSSPP's controls? e.g location of .cfg file as the buttons on my Xbox 360 controller are messed up for it.

closed 2015-10-30 by robertybob

HerbFargus, 2015-10-30:
That would be me. the configs (at least the diagram on the wiki) correspond to the libretro build. I used the default settings like it would for the snes. Granted I don't know the default core configs so that was my best guess. if thats incorrect then I can update it.

HerbFargus, 2015-10-30:
the mappings are fine on my snes controller, and they worked for ronoh as well. Note that the default diagrams for the libretro configurations are all dependent on the user mapping their controllers based on the diagrams on this page: https://github.com/RetroPie/RetroPie-Setup/wiki/First-Installation#configurations otherwise they won't match up. (also many emulators can have their controls changed through core input remapping options in the rgui which also will make the default diagrams obsolete.)

robertybob, 2015-10-30:
@HerbFargus Sorry I should have been clearer, it's how to map the controls for the non Libretro PPSSPP I'm after - if it really performs much better than the Libretro one :) e.g. Mupen64Plus has the /opt/retropie/configs/n64/mupen64plus.cfg file and I wonderd if there was a PPSSPP equivilant (?). The wiki says to open the Settings >> Controls >> Control Mapping menu but how do I get to that?

joolswills, 2015-10-30:
press escape ?

robertybob, 2015-10-30:
Thanks! I don't have a keyboard in my setup so didn't think to do that.

joolswills, 2015-10-30:
You can't expect to mess around with stuff in the experimental section without one - get a bluetooth keyboard or something :)