RetroPie/RetroPie-Setup #1070: RPi 1 RetroArch binary for Jessie not working

diogomsantos, 2015-10-30:
The RPi1 RetroArch binary for Jessie isn't working.

Here is /tmp/runcommand.log:
/opt/retropie/emulators/retroarch/bin/retroarch: error while loading shared libraries: libwayland-egl.so.1: cannot open shared object file: No such file or directory

From source works ok.

joolswills, 2015-10-30:
Probably due to retroarch building with some additional features/dependencies on my build machine due to differing installed packages. Thanks. Will check and modify configure flags accordingly.

joolswills, 2015-10-30:
aah . the egl issue again. I had the debian egl installed when it was built, which causes stuff to build against that rather than the /opt/vc egl libraries. This affected some other stuff too. I will rebuilt and upload a new binary now.

joolswills, 2015-10-30:
new binaries built. thanks.

closed 2015-10-30 by joolswills