RetroPie/RetroPie-Setup #1071: attact-mode front-end as an alternative to emulationstation

objectchris, 2015-10-30:
As per http://blog.petrockblock.com/forums/topic/suggestion-attact-mode-front-end-as-an-alternative-to-emulationstation/#post-108794

emulationstation development seems to be in hiatus at the moment, but check this out: http://attractmode.org/ – a themeable umulator frontend, that works on the pi!

it seems somewhat more ‘busy’ than emulation station (more like hyperspin), so probably better for more curated systems, so i we wouldn't want to do completely get rid of emulationstation (which is great for large libaries). but it would be awesome to get this system as an alternative! I think it would slot in quite well with the rest of the architecture, plus it’s still in development!

tipoto, 2015-11-04:
Here are the instructions to compile it on the Raspberry Pi:
https://github.com/mickelson/attract/wiki/Compiling-on-the-Raspberry-Pi-%28Rasbian-Wheezy%29

HerbFargus, 2015-12-04:
This is the correct page: https://github.com/mickelson/attract/wiki/Compiling-on-the-Raspberry-Pi-%28Raspbian-Wheezy%29

HerbFargus, 2016-02-09:
Just as a reference I've tested the compilation instructions on Jessie and they don't work. may need to be ammended for jessie.

tipoto, 2016-02-09:
I'm now using an Odroid XU4 with a Debian Jessie based image (OGST), it's been very complicated to get it compiled on it, but after struggling a lot and with the help of a few persons we eventually managed to do it.
Here is the "How to compile" thread I created, these instructions won't work directly on the pi without any modifications of course, but hopefully the problems are similar enough to give some ideas:
http://forum.attractmode.org/index.php?topic=523.msg3671#msg3671
This method, at some point, use a patch, you can take a look at it to see the content.

HerbFargus, 2016-02-15:
Based on your changes @tipoto I've gotten it to compile successfully and I've added a wiki page for it: https://github.com/mickelson/attract/wiki/Compiling-on-the-Raspberry-Pi-(Raspbian-Jessie) 

Out of curiosity it necessary to have an xserver of sorts in order to run it? or would it be possible to run directly from the framebuffer like emulationstation does?

tipoto, 2016-02-15:
> Out of curiosity it necessary to have an xserver of sorts in order to run it? or would it be possible to run directly from the framebuffer like emulationstation does?

I can't answer since I don't have enough technical knowledge, but the RPi has this great ability to be able to open and close x11 at will, very easily, by using the "xinit" command.
When I was using the RPi2 with Attract-Mode, I could run advmame 0.94/1.2 and even OpenBOR in frambuffer mode, directly from the AM interface, but I had to create scripts that were closing AM before launching the game then re-opening it after the game was exited. Of course it would be easier if AM could be run directly in fbdev, but then you would have to run x11 for the apps or emulators or port games that can't be compiled for fbdev.

joolswills, 2016-02-15:
very few of the games / emulators we ship with require X - so it is preferable to not require X for the launcher - I will look into this when I get some time.

Devil78, 2016-02-15:
I've made a custom menu in Emulation Station used, moreover, to lauch Attract Mode. 
[image]
[image]

robertybob, 2016-02-16:
@Devil78 using the new changes in 3.5, would it theoretically be possible to launch Attract Mode directly from EmulationStation without the need for a sub-menu (like Kodi can be) ?

HerbFargus, 2016-03-13:
Mehstation may also be worth considering as a frontend (much simpler than attractmode)

https://github.com/remeh/mehstation

robertybob, 2016-03-13:
Did you see the Reddit post about v 1.0? lol. Looks very promising indeed!

joolswills, 2016-03-26:
As it requires X to run, and most of the applications on retropie run directly on the framebuffer it's not going to be too useful to us and it will be a pain to script to working around that on top of our existing system I should think.

HerbFargus, 2016-03-26:
I agree. If we were to look at another frontend mehstation looks to be more promising.

dankcushions, 2016-03-30:
Should I raise our concerns on the attract-mode tracker, in case this might be changeable? I don't know enough about the ramifications, but is it feasible that attract-mode could be written without X11? I guess emulationstation uses SDL instead? And we don't want X11 because none of our packaged emulators use it (in fact, it has to be disabled when they're running(?)).

I think given how popular Retropie is, it might be worth me raising it with them, just in case there's any interest in widening their reach. I still like the idea of a hyperspin-type interface for curated playlists, plus that project is still busy.

HerbFargus, 2016-03-30:
It uses sfml which would require a massive amount of work to make if run without X. It wouldn't be worth his time.

http://stackoverflow.com/questions/29620061/starting-a-graphical-application-using-sfml-on-raspberry-pi-boot

You could try suggesting it but I don't think it's feasible.

Palatosino, 2016-04-15:
This frontend are very beautiul!

I hope that, is there is no problem to install, the scripts give to us the possibility to install different frontends:

1. Install with EmulationStation

1. d

Install with Mehstation (will install X11))

remeh, 2016-04-28:
Hello! mehstation developer here.
I'm very glad to see mehstation appear as a possible alternative to ES in RetroPie!
The main difficulty to integrate mehstation in RetroPie would be IMO that the configuration of the emulators/roms/... is currently done in another graphical app (mehstation-config) and not using a "discover roms in directory" behavior. But this is something I'm completely ready to work on if it could help to integrate with RetroPie.

Finally I want to confirm that mehstation can definitely run without X11 as I use SDL2 for the rendering (and I've already setup a Rasp using mehstation without X11).

dankcushions, 2016-05-02:
attractmode doesn't need X anymore! https://github.com/mickelson/sfml-pi/commit/3330dd982060abee8b8f69ffe77c0610d6e94da4

Floob has a retropie build here: https://retropie.org.uk/forum/topic/93/attract-mode-with-retropie-alternative-to-emulationstation/137

Redemp, 2016-05-02:
@remeh

That would be so sweet! Finally we might have to new frontends for RetroPie. 

One for the Arcade Cabs or just that arcade feel in the living room. 

One for the console experience. 

mehstation is more up my alley and with favorites, Hide Game, Kidgame, Kid/Kios mode I'm sold!

joolswills, 2016-11-22:
attractmode is now available in experimental. It currently will try and scan / add all the games in your collection on install - simple gui for doing that coming later. Working on mehstation now (almost done - directly creating the gameslists currently via sqlite, but if the frontend can "scan" itself this wont be needed)

Redemp, 2016-11-22:
@joolswills Shut the front door!

joolswills, 2016-11-22:
Mehstation is available in a PR - https://github.com/RetroPie/RetroPie-Setup/pull/1760 but some problems currently which I have detailed. Perhaps @remeh can advise - it could be our fault for the way we construct an initial db or related to mehstation auto scanning not being complete in some way. 

Would be nice if mehstation had some commandline operations to avoid manual construction of the database / creation of the systems via sqlite3

mickelson, 2016-11-23:
Hello.

I'm the author of attract-mode.  Thanks for adding this to retropie!  Let me know if there is anything that I can do upstream to assist...

thomas-mc-work, 2016-12-05:
Are the experimental mehstation packages ready for an early end user test? I was able to comipile and install it. But then it failed when running:

    $ mehstation 
    ** Message: Found the resources dir: /opt/retropie/supplementary/mehstation/share/res
    ** Message: Found configuration file in /home/thomas/.config/mehstation/mehstation.conf
    error: XDG_RUNTIME_DIR not set in the environment.

    ** (mehstation:7962): CRITICAL **: Can't init the SDL: No available video device
    /usr/bin/mehstation: line 3:  7962 Floating point exception"/opt/retropie/supplementary/mehstation/bin/mehstation" "$@"
    ~

(I'm on an up2date debian 8)

joolswills, 2016-12-05:
@thomas-mc-work yes - please use the retropie forum for support https://retropie.org.uk/forum/

joolswills, 2016-12-05:
@mickelson thanks. I added a small feature request for the ability to add/remove/update emulators/displays from commandline.

DavidJ1975, 2016-12-09:
Love the look of the new attract mode, but I've been failing on the upgrade. I'm using berryboot and the install provided in the OS selection. I've tred multi times using info in videos and both versions (Jessie and Wheezy) in the mickelson wiki without success. Has anyone successfully done this with a pi3 and berryboot? Have a link or instructions?

stevetb, 2016-12-14:
@mickelson I think we are all looking forward to Attract Mode on Retropie.  I have only used it a little bit.  Does it have scraping built in?  If so, does it scrape against EmuMovies?

joolswills, 2016-12-14:
@DavidJ1975 Please use the RetroPie forum for support.

@stevetb for help with using attractmode you can use the retropie forum or the attract mode forum - http://forum.attractmode.org/

this is not a support area but a ticket for implementing the frontend into RetroPie (which has been done - at least preliminarily). I will close this issue now.

closed 2016-12-14 by joolswills