RetroPie/RetroPie-Setup #1072: RetroArch hotkey conflicts (FBA, MAME, etc.)

HerbFargus, 2015-10-31:
As it's unlikely retroarch will fix this, just opening an official ticket for discussion to get the hotkey issue sorted on retropie's end:

see post:
https://github.com/libretro/RetroArch/issues/2230

joolswills, 2015-11-01:
I posted a patch in the thread above - just a heads up for @gizmo98 so we don't duplicate work - feel free to improve on it - it's not very pretty! :)

gizmo98, 2015-11-01:
Cool. I know the part you are modifying ;)
https://github.com/libretro/RetroArch/pull/1311

gizmo98, 2016-02-20:
Is there a unofficial fix?
https://github.com/recalbox/recalbox-buildroot/pull/72

objectchris, 2016-02-20:
that removes the check entirely. we could patch it out like that as a last resort. I think jools' fix at the end of https://github.com/libretro/RetroArch/issues/2230 was the right approach for upstream but it needs debugging. was going to try that today actually!

btw the code for this is now in input_driver.c not runloop.c

joolswills, 2016-02-20:
well,, my fix doesn't work right. so we could use this fix with a timer (blocks after 10 frames).

joolswills, 2016-02-20:
This fix is better than disabling as if you have it on a spare key, you can hold it for 10 frames, and it will block keys from being sent to the core as you would want.

objectchris, 2016-02-20:
I was looking at the wrong commit :( yeah the timing one seems like it would be upstream friendly also. I like it!

joolswills, 2016-02-20:
might be worth asking if upstream would consider it before we include it - they may well not, but worth an ask.

joolswills, 2016-02-20:
Although the hotkey blocking function returning the count and doing the check outside - I wonder if the timing check should be done in the function itself so it still returns a boolean. Would seem more logical that way.

objectchris, 2016-02-20:
something like this? 2ad8321

i'm checking my fix on the pi now. Works on my macbook ok I think. https://github.com/objectchris/RetroArch/commit/2ad83219211f072422a176546b95fe18ed0ddd86

joolswills, 2016-02-20:
That looks more complicated -:)  I was thinking to make it simpler by just doing `return  hotkey_counter > HOTKEY_DELAY`

joolswills, 2016-02-20:
well saves the addition of another variable anyway.

joolswills, 2016-02-20:
also hotkey_counter should not be incremented if it's already greater than HOTKEY_DELAY

objectchris, 2016-02-20:
https://github.com/objectchris/RetroArch/commit/0158c13949f5e9212e0a181050d8df141afa93c9 removes extra variable - this is how initially did it but wondered if it was still readable?

joolswills, 2016-02-20:
you dont need the "use_hotkey_enable && enable_hotkey" check again.

objectchris, 2016-02-20:
works on pi fine. i'm sort of tempted to reduce the count to 5 though. my natural hotkey pressing combos were faster than 10 frames so it was letting the second key input register in game. i'll do those changes now, ta :)

objectchris, 2016-02-20:
````
   if (use_hotkey_enable && enable_hotkey)
      hotkey_counter++;
   else
      hotkey_counter = 0;
   
   /* If we hold ENABLE_HOTKEY button, block all libretro input to allow
    * hotkeys to be bound to same keys as RetroPad. */
   return (use_hotkey_enable && enable_hotkey && (hotkey_counter > HOTKEY_DELAY));
```
i don't like removing the second use_hotkey_enable && enable_hotkey as i think it makes it harder to read and the comments would need rewording. how about:

````
   /* Increment frame counter. Only declare hotkey in use if held for
    * HOTKEY_DELAY frames, so its native function is initially
    * passed to core */
   if (use_hotkey_enable && enable_hotkey && (hotkey_counter < HOTKEY_DELAY)) {
      hotkey_counter++;
      if (hotkey_counter > HOTKEY_DELAY)
         use_hotkey_enable = false;
   }
   else
      hotkey_counter = 0;
   
   /* If we hold ENABLE_HOTKEY button, block all libretro input to allow
    * hotkeys to be bound to same keys as RetroPad. */
   return (use_hotkey_enable && enable_hotkey);
````

new to C. probably a better way of handling this :)

joolswills, 2016-02-20:
you may not like removing the second check, but it is wrong.

Sorry, but you have made this a lot worse than the original code imho.

joolswills, 2016-02-20:
I dont really want to go back and forth with this as the time I am spending I could have just written the changes - I will include a patch for RetroPie if others are happy and I will post it on the RetroArch ticket to see if they are interested.

objectchris, 2016-02-20:
perhaps you should have done that. i was only trying to get things moving. i look forward to the patch.

joolswills, 2016-02-20:
I appreciate you are trying to help. But you did ask me and so I was giving my thoughts, the original patch was good - I was only suggesting a one line change. Rather than trying to work around the comments, just change the comment :)

joolswills, 2016-02-20:
Testing this https://github.com/joolswills/RetroArch/commit/31771d930ba813e0c03cd1ef0b154987eb681ed6

objectchris, 2016-02-20:
same as my last fiddle now :) https://github.com/objectchris/RetroArch/commit/72e2ee3d28a4297673293b6982fcb107538a677c
i updated the comments a bit and i think it reads ok. feel free to nab them.

tested fine for me. 5ms seems about right.

joolswills, 2016-02-20:
on a 50hz refresh this would be 83ms or so I think.

joolswills, 2016-02-20:
not sure this is actually working correctly - seems to trigger early, but maybe it's me - anyway. I will look further into it later - no rush.

joolswills, 2016-02-20:
Yeh it's just me actually I think :)

joolswills, 2016-02-20:
so If @HerbFargus @petrockblog and @gizmo98 are happy I will include my patch in our retroarch build.

joolswills, 2016-02-20:
s/my/the patch

joolswills, 2016-02-20:
might have helped if I actually applied the changes over the latest upstream code. forgot to merge.

joolswills, 2016-02-20:
https://github.com/RetroPie/RetroPie-Setup/pull/1278

HerbFargus, 2016-02-22:
This should be resolved as of https://github.com/RetroPie/RetroPie-Setup/commit/ccb3f73b917f58ef69e73e746bdaaea83772bdb4 Will reopen if issues are found, but from my tests it is a much better solution than what we have had.

closed 2016-02-22 by HerbFargus