RetroPie/RetroPie-Setup #1073: PS3 controller pairing script

tiagojpdias, 2015-11-01:
Hey,

It seems that the script which handles pairing PS3 controllers with bluetooth dongle its not quite right.

It lacks the 3 steps, like
1 - "Connect the USB dongle"
2 - "Connect controller by USB"
3 - "Disconnect USB cable and press PS button to pair"

I found where the possible errors were made by cherrypicking this commit: 5182170d6f7f90beb07e44ad10887599eaffe1a0

Here the script behaves like it should and I successfully pair the controller.
If any more info is needed let me know.

Cheers,

gizmo98, 2015-11-01:
It's intented. Pairing works anytime if you have retropie 3.2.1. If you use an older SD card image update setup script, remove and reinstall ps3 controller driver.

gizmo98, 2015-11-01:
If it is still not working please post your hardware setup.

tiagojpdias, 2015-11-02:
Hi @gizmo98,

I've managed to setup the controller before reporting the issue.
I'll try to make a fresh install of my system and then trying to replicate it (because I don't want to "undo" all the work done to this point)

I'll post the hardware setup just to be documented:

* Raspberry PI B (Raspbian Jessie with kernel 4.1.7+)
```bash
pi@rpi ~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
pi@rpi ~ $ uname -a
Linux rpi 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l GNU/Linux
```
* Last pull
```bash
pi@rpi ~/RetroPie-Setup $ git show | grep Date
Date:   Sat Oct 31 15:34:21 2015 +0000
```
* Bluetooth:
```bash
pi@rpi ~ $ lsusb | grep Blue
Bus 001 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```

* PS3 sixaxis controller
```bash
pi@rpi ~ $ dmesg | grep Sony
[   58.801287] input: Sony Computer Entertainment Wireless Controller as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.0/bluetooth/hci0/hci0:1/0005:054C:0268.0001/input/input3
[   58.805907] sony 0005:054C:0268.0001: input,hidraw0: BLUETOOTH HID v1.00 Joystick [Sony Computer Entertainment Wireless Controller] on 00:15:83:0c:bf:eb
```

If you need any more information let me know,

Cheers

gizmo98, 2015-11-02:
I have not tested raspbian jessie until now. It could be systemd, a newer bluez package or gcc-4.9 brake compatibility. I will take a look later.

joolswills, 2015-11-02:
I wouldn't recommend running retropie on jessie yet - it's still somewhat a WIP, as there are other configuration problems too (need to rework some other bluetooth functionality and some other things)

joolswills, 2015-11-02:
from the README.,md

>It has some support for Raspbian Jessie / OSMC but requires building from source - and there may be problems with modules."

gizmo98, 2015-11-22:
@tiagojpdias 
Please update RetroPie-Setup and reinstall PS3 driver. I have not tested raspbian jessie but issues i had last time i tested should be solved now.

gizmo98, 2015-12-23:
This issue is fixed. Tested with retropie 3.3 jessie beta.

closed 2015-12-23 by gizmo98