RetroPie/RetroPie-Setup #1077: Enabling option to configure Retroarch controller

biscuits99, 2015-11-04:
I know that recently the option to configure a retroarch controller has been removed to avoid confusion that it may be required now that ES handles the config file generation:
https://github.com/RetroPie/RetroPie-Setup/commit/6d7d74cd781af3537b190ecac022bd29938f1d03

However, I've found some instances where the ES generation doesnt always match the old style of generation. An example is with the 8bitdo controllers, there are mismatches on some of the buttons although the axis are fine. This is referred to here in step 5 and 6:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

I think this user *may* also be having the same issues with the non-descript controller:
http://blog.petrockblock.com/forums/topic/neogeo-exit-button-not-working/#post-109124

Would it be possible to re-instate this menu option, albeit with a warning that it shouldnt be required unless you know what you are doing type message?
I'm happy to provide the retroarch cfg file with ES and the old file for comparison if that helps?

biscuits99, 2015-11-04:
Different button mapping shown here:

http://pastebin.com/raw.php?i=zpJaMY7P

joolswills, 2015-11-05:
It was only removed from the ES menu - it's still available from retropie-setup. The problem was people kept using it, and usually it isn't needed.

I would need the es_temporaryinput.cfg file too which contains the full config generated by ES sent to our input configuration tool - and if you know which bits are generated wrong, it will save me time from comparing each config line etc :)

joolswills, 2015-11-05:
updated above post.

joolswills, 2015-11-05:
based on the configs above - I can see that the ID for button B is different - I cannot see how that can happen unless a different button is pressed for B in ES vs what is pressed for B in retroarch-joyconfig.

joolswills, 2015-11-05:
(I see a lot of the button ids are higher - so there may be something odd at play regarding how ES handles the mapping vs ES, but very strange). The controllers were connected in exactly the same way / in the same mode in both cases ? what does button B output from jstest ?

joolswills, 2015-11-05:
Also, does it make a difference if you switch the input driver in retroarch from udev to SDL2 (or linuxraw) ?

joolswills, 2015-11-05:
Sorry for ticket spam - I have updated the above message so please re-read thread.

biscuits99, 2015-11-05:
Ah, I see - my mistake, I thought it was being removed from the retropie-setup. Ok, no real problem then, at least there is a fallback option.

I have put the es_temporaryinput.cfg here:
http://pastebin.com/raw.php?i=ePRFDFi3

It appears as though all buttons are wrong, and all directions are fine.

jstest is reporting button B as "1". Which doesnt work "in-game" but "12" does, which is what retroarch-joyconfig generates.

Yep, the controller was connected in exactly the same way for both sets of controller config files, no difference at all.

To test the input driver, is it valid just to change the line to
input_driver = "sdl2" ?

joolswills, 2015-11-05:
I don't actually know for sure but please try changing that yes. It looks like in the case of your controller, via the udev driver, different button codes are used.

biscuits99, 2015-11-13:
I've just tried setting it to sdl2 in the controller file, and the only button recognised is the 'a' button which is acting as the start button.

What should I change these to in the /all/retroarch.cfg?  

    # Input driver. Depending on video driver, it might force a different input driver.
    # input_driver = sdl

    # Input device driver. (Valid: linuxraw, sdl, dinput)
    # input_joypad_driver =

biscuits99, 2015-11-13:
Just changed this  

    # input_joypad_driver =
to  

    input_joypad_driver = linuxraw

All buttons working fine in  game. But no D-pad directions work at all.

biscuits99, 2015-11-13:
Changed both to SDL2.
Button and directions working! WooHoo!

One thing to also add is that when setting the joypad up in ES, it ignores the L2 and R2 buttons, whereas it works/detected via the retroarch setup.

joolswills, 2015-11-13:
I suspect this controller has some firmware problems - no idea why the l2/r2 presses would not be detected by sdl2. Good to know it works like that. It's worth noting that retroarch-joyconfig has been removed from retroarch now, so that is no longer a viable option really.

joolswills, 2015-11-13:
@gizmo98 @petrockblog  due to this controller (and possibly others) producing different button ids with sdl2 vs udev, it might be worth us switching retroarch to sdl2 and use sdl2 for the configs. It needs testing though in case there are downsides to using sdl2 for input.

gizmo98, 2015-11-14:
The main difference between linuxraw and udev was udev supports hot-plugging. If retroarch is running and you plug-in or connect wireless a controller, retroarch recognizes the controller instantly. If you use linuxraw it is necessary to restart retroarch before you can use a new controller. It's a handy feature if you have some bluetooth controllers and want to play multiplayer games. Without this feature it is necessary to connect "all" controllers before you start a game.

If retroarchs's sdl2 driver supports hot-plugging i would suggest switching to this driver.

joolswills, 2015-11-14:
@biscuits99 if you leave it as "udev" in the controller file, but set the main retrooarch.cfg to sdl2, what does it do then - just want to confirm if it needs sdl2 in both configs, or not.

biscuits99, 2015-11-14:
With this line in the controller file  

    input_driver = "udev"
and this line in the main cfg  

    input_joypad_driver = sdl2
it works fine still, no issues.

I'm not sure of the signifcance of this line in the main cfg  

    # input_driver = sdl

Also, I notice after navigating ES menus for a while then launching a game, I often see this line repeated after the runcommand prompt  

    Info: The key you just pressed is not recognized by SDL. To help get this fixed, please report this to the SDL mailing list <sdl@libsdl.org> [VDEV Keycode 304

It seems to just be whatever Keycode 304 and 305 are.

joolswills, 2015-11-15:
Thanks - in regards to the error message I guess I would do as it asks and report it to sdl :)

HerbFargus, 2015-11-23:
I'm pretty sure 304 = LSHIFT and 305 = RCTRL

gizmo98, 2015-11-28:
The recalbox guys have an interesting sdl2 related commit:
https://github.com/recalbox/recalbox-buildroot/commit/91d0ff6ad573051e780ab8800208d8890dbb4b1f

joolswills, 2016-10-22:
Closing this as the OP is solved - joypads can be manually configured inside RGUI if needed I believe.

Regarding the sdl2 commit in recalbox, I suspect it will introduce issues with some emulators, but perhaps it's worth testing - a new ticket can be opened if needed.

closed 2016-10-22 by joolswills

dikdom, 2018-10-20:
Same issue here with retroPie v4.4, and with 2pcs of cheap (~10 USD) chinese bluetooth controllers. 
Configured properly with EmulationStation. Navigation works, buttons work in ES. In lr games (SNES, NES, GBA, etc...) the navigation works, buttons don't.
jstest /dev/input/js[01] prints for button A, B -> 0, 1 etc..., same values as configured in the controller's config file in /opt/retropie/configs/all/retroarch-joypads/Gamepad.cfg file.
Using the input configuration in retroarch, it prints different values for the buttons. This is the reason for not working buttons.. (spice in the story that even the two gamepads have different button values for some buttons also - in retroarch input config screen)
Finally I edited the retroarch.cfg file, entering sdl2:
>
# Joypad driver. (Valid: linuxraw, sdl, dinput, udev)
input_joypad_driver = "sdl2"
>
(please note the valid options, it's sdl instead of the used sdl2!)
Long story short: autoconfiguration works, hotplug works, DPad, buttons work.
Coolness, happiness.

PR #3724 (2023-06-14): amiberry: update to 5.6.1
Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation. Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #108 [... cut]
commit 042ae93c (2023-06-14): amiberry: update to 5.6.1

Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation.
Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #10 [... cut]