RetroPie/RetroPie-Setup #1078: PPSSPP Improper Controller Keys

Omnija, 2015-11-05:
Recently installed ppsspp and tried to setup my controller, but when i pressed (B) it would get Direction (Up) and (A) would be Directional (Left). The Controller inputs are fully incorrect. 
I'm using an xbox 360 rock candy controller.

joolswills, 2015-11-05:
Maybe they have hardcoded mapping - You should report this upstream perhaps - maybe by first starting on their forum if it's a configuration issue - http://forums.ppsspp.org - nothing I can do to help (don't even have your controller). try another controller too ?

joolswills, 2015-11-05:
There is also the libretro core ?

Sorry to close, but I don't think there is anything we can do currently - seeing if other users have the problem on the forum, and if they managed to remap their controllers may be an option. Or use a keyboard? :)

closed 2015-11-05 by joolswills