RetroPie/RetroPie-Setup #1080: [lr-fba] new release - v0.2.97.37

nemo93, 2015-11-08:
Hi!! seems that good fellas from FBA just released a new version with bugfixes, new drivers, etc. (read here http://www.fbalpha.com/view/227/). And @barbudreadmon was kind enough to update his libretro port (https://github.com/libretro/libretro-fba). 

I'm building it right now and will try to update this thread to let you know if it's worth it! 

Thanks :)

joolswills, 2015-11-08:
Yeh saw that yesterday - not built it yet, so am interested in feedback.

nemo93, 2015-11-08:
Stable, no crash during a 2h play session. Brings lot more options in RetroArch Core Options especially for Neogeo or other Arcade systems. CPS3 run lot better one this one as Jojo games have no more glitches and Red Earth just works this time. Don't see any difference on Neogeo games as previous version was already great. Also it seems this newer version handles a lot better some Sega games as Alien Storm works fullspeed and without glitches. Same goes with Turbo Outrun. 

At the end I'd say it's a no brainer and definitely worth the upgrade.

barbudreadmon, 2015-11-08:
Sorry, i still have no clue about the issue with irem games on arm, what i know : 
- It is not endian related
- It is not x86 specific instruction related (no ASM code)
- It is probably related to nec V30 cpu (all affected games uses this cpu)
- The cpu emulator is not necessarily the culprit since m90 emulation (which also uses V30) is working

joolswills, 2015-11-08:
I thought it might be due to ARM having unsigned chars by default, but it made no difference when I forced it to signed-char :/

nemo93, 2015-12-22:
been added to the 3.3 release \o/ (https://github.com/RetroPie/RetroPie-Setup/wiki/Changelogs)

closed 2015-12-22 by nemo93

dips52, 2016-11-09:
nemo93 

Can you give any more information on how you got red earth to run? I have tried what feels like all versions but still have no luck getting it to run on latest FBA with pi3. 

I'm trying this one again but don't have a good feeling about it: 

Capcom Play System 3 Emulator v1.0 - "Warzard / Red Earth" - année: 1996 - fabricant: "Capcom" 

Télécharger : warzard.zip 
Taille : 34.51 Mo

PR #3724 (2023-06-14): amiberry: update to 5.6.1
Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation. Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #108 [... cut]
commit 042ae93c (2023-06-14): amiberry: update to 5.6.1

Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation.
Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #10 [... cut]