RetroPie/RetroPie-Setup #1081: [RPi2 Retropie 3.2.1] Error installing lr-mame2010: "mame2010_libretro.so not found"

kilateped, 2015-11-11:
Hello, hope you can help me.
I'm trying to install experimental lr-mame2010 package trough RetroPie setup, but I can't. It says it couldn't find mame2010_libretro.so in tmp/build/lr_mame2010.
What am I doing wrong?

Thank you.

joolswills, 2015-11-11:
did you update retropie-setup script first ?

kilateped, 2015-11-11:
Yes I did.

joolswills, 2015-11-11:
then I will need a log from the RetroPie-Setup/logs folder - use a site like pastebin.com or similar.

kilateped, 2015-11-11:
http://pastebin.com/697kmBUH
Thanks.

joolswills, 2015-11-11:
You have not updated the retropie-setup script

kilateped, 2015-11-11:
How many times do I need to update it? First time I entered retropie-setup script I chose "Update RetroPie- Setup script" and entered ok.
Do I need to update it every time I want to install a package? Or maybe I didn´t update correctly?

joolswills, 2015-11-11:
it should show an error if it didnt update correctly. try "git pull" from terminal also.

lr-mame2010 was fixed on the 6th November (after upstream changes). So long as you updated after that lr-mame2010 should built correctly. I can see from the log it is referencing stuff that has changed so it must be it didn't update.

joolswills, 2015-11-11:
If you mean you updated retropie-setup at some point some weeks ago, then you will need to update it again.

kilateped, 2015-11-11:
This is from the first time I tried to install:
http://pastebin.com/7h2Zm0gN
I did some more tries without any success, then I replace lr-mame2010.sh file with the one prior 6th November, and tried again without success. The first log I send you is from this last try.

joolswills, 2015-11-11:
Why are you manually replacing files?

That log file is also an out of date lr-mame2010.sh

Just update RetroPie-Setup (if you have modified local files, you will need to `git reset --hard` or `git stash` your changes or else it won't be able to update).

kilateped, 2015-11-11:
Ok. I´ll start from scratch tonight, format SD card, Install RetroPie image, update setup script and install lr-mame2010. I´ll tell you if I am successful.
Thank you for your help. I´ll tell you later.

joolswills, 2015-11-11:
it works as I just built it here. Just follow my instructions above and you should be able to sort it. No need to start from scratch.

kilateped, 2015-11-11:
Ok.

kilateped, 2015-11-11:
Ok, did what you said and success ;) 
Thank you very much!
I found some games running slowly (like run and gun 2, versus net soccer...) do you think it's possible to get them run near normal speed? Maybe overclocking or is just the mame version not capable of running them smoothly?

joolswills, 2015-11-11:
there is a reason the mame2010 is in experimental. the pi2 cpu is too weak i'm afraid. best bet is the 2003 mame perhaps.

joolswills, 2015-11-11:
You can use the forum for further support / feedback.

PR #3724 (2023-06-14): amiberry: update to 5.6.1
Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation. Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #108 [... cut]
commit 042ae93c (2023-06-14): amiberry: update to 5.6.1

Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation.
Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #10 [... cut]
closed 2015-11-11 by joolswills