RetroPie/RetroPie-Setup #1082: Please remove limelight from experimental packages!

Haris1977, 2015-11-11:
As the author answered me here (https://github.com/stsfin/RetropieLimelightInstaller/issues/1#issuecomment-155789338) he 's lost interest in the project:(

On the other hand there is an updated version here (https://github.com/irtimmer/moonlight-embedded) called moonlight embedded (uses c code instead of java). It works pretty well even though i haven't been able to incorporate it in my retropie image yet (i did use the raspbian wheezy version with success though) 

So, if you could add the latter (instead of the first one), it would be great!

diogomsantos, 2015-11-11:
Yeah, moonlight would be a great addition to RetroPie! I would help in the script, but I don't have a compatible nvidia card at the moment...

cbriantwizcorp, 2015-11-12:
Yes, fully agreed, I am also desperate to have a working limelight/moonlight straight from Retropie!

Haris1977, 2015-11-12:
Well. I ve managed to install moonlight embedded and adding a shortcut in ES as well in my pi1!!. If anyone wants here are the steps (commands from ssh). There is no need in compiling from source or adding dependancies. SSh and copy - paste the above 8 commands:

1) sudo apt-get update
2) sudo apt-get dist-upgrade
3) sudo nano /etc/apt/sources.list  and add the line below:

deb http://archive.itimmer.nl/raspbian/moonlight wheezy main   

then save and exit (control+x, yes)

4) sudo apt-get update
5) sudo apt-get install moonlight-embedded      (Press ‘y’ in case of any questions)
6) mkdir ~/Moonlight
7) cd ~/Moonlight
8) moonlight pair XXX.XXX.X.XXX (your pc ip) and follow the onscreen instructions with the 4 pin digit.

Now you have paired your pc. Then you’ll need to make a mapping file for your controller.  After you type the command it will ask you to press all the buttons in a specific order, so it is better to use these commands from your pi and not from ssh (quit emulationstation and use the commands) .

    cd /opt/retropie/configs
    mkdir moonlight
    cd moonlight
    moonlight map controllername.map

Now to add a shortcut to your pc in Emulation Station do the following:

    sudo cp /etc/emulationstation/es_systems.cfg es_systems.cfg.bak
    sudo nano /etc/emulationstation/es_systems.cfg

Add the following to the very bottom of your config before </systemList>:

<system>
  <fullname>Steam</fullname>
  <name>Steam</name>
  <path>~/RetroPie/roms/moonlight</path>
  <extension>.sh .SH</extension>
  <command>%ROM%</command>
  <platform>steam</platform>
  <theme>steam</theme>
</system>

Save your config and do the following:

    mkdir ~/RetroPie/roms/moonlight

Now we’re going to create a script or multiple scripts for starting moonlight.  I have 3 different scripts that connect to the same computer with either 1080p/30fps, 720p/60fps, or 720p/30fps so I can stream in different qualities depending on how well a certain game streams.

    cd ~/RetroPie/roms/moonlight
    sudo nano 1080p30fps.sh

Then paste in the following (adjust for your computer’s ip address):

#!/bin/bash
moonlight stream 192.168.0.101 -1080 -30fps -mapping /opt/retropie/configs/moonlight/controllername.map

 Save that file and then make it executable.

    sudo chmod +x 1080p30fps.sh

There is also a Steam theme so that Steam matches the other platforms in EmulationStation. Simply download the zip from here http://jordanhotmann.com/wp-content/uploads/2015/09/steam.zip, extract the steam folder, and copy it to /etc/emulationstation/themes/simple for bonus beautification.

***Be carefull: It only works for simple theme and not in carbon, so you must download first simple theme from retropie menu!!***

If you cannot copy steam folder (because of permission errors), first copy it to /home/pi/ with filezilla and then use the below command with ssh:

sudo cp -r /home/pi/steam /etc/emulationstation/themes/simple

Restart you Pi (or just Emulation Station) and you should see Steam in your console list.  Now sit back and enjoy all your favorite PC games on your couch!!

All credits/thanks goes to these sites and to my thoughts that i used to compose all of them together:

http://jordanhotmann.com/guide-install-moonlight-on-raspbianretropie/
https://github.com/irtimmer/moonlight-embedded

So, as a last thing to do, is someone who can put all of these together in a script through experimental packages !!!

cbriantwizcorp, 2015-11-12:
Awesome!!! Big thanks @Haris1977, I will try that as soon as I can !! :)

amivit, 2016-01-25:
You are the man Haris1977! Thank you so much, I will be working on this tonight.

djrockgt, 2016-02-11:
I _just_ started working on a project exactly like this. I'm using pancakeykakes's image file for the base install. It includes ES, RetroPie, and Kodi along with some basic configurations: https://tripi.junaos.com/

From here I followed something similar to the above directions by Haris1977. Those commands were found in the following article. I updated the commands so that it would pull the latest version of moonlight: http://jordanhotmann.com/guide-install-moonlight-on-raspbianretropie/

Now I'm working on installing a VPN client so that I can do streaming from anywhere. The VPN server is SoftEther and I run it on my gaming PC. It is open-source and VERY fast: https://www.softether.org/

You can find the thread with the latest moonlight commands and all the things I'm experimenting with on this subreddit: https://www.reddit.com/r/raspberry_pi/comments/3qi3tz/tripi40_is_now_available_as_a_bootable_image_and/czp5wwv

Hope this helps someone. A shell script would be great but I'm completely new to Linux as well. I literally learned how to wget three days ago.

djrockgt, 2016-02-12:
I finally got it working. Used OpenVPN and made sure it started at boot. This allows me to do moonlight streaming from anywhere.

I pulled from a number of different resources but basically I used SoftEther to set up the VPN server on my PC, had it autogenerate the OpenVPN config file from it, and then imported it into the Pi. You then create an authentication text file and add your username and password on separate lines then save that into the openvpn folder (a file named "user.txt" for example). Open the .ovpn config file and add the name of your text file after auth-user-pass (e.g. auth-user-pass /etc/openvpn/user.txt). Finally, change the extension of the .ovpn file to .conf to have it auto-start on boot.

This is the quickest way to do this without having to build ovpn configs from scratch.

amivit, 2016-02-13:
So in that scenario you are just using the pi as VPN gateway, and moonlight from your client to the desktop through the tunnel? Very cool, I wish I had the upload speed to try the same.

djrockgt, 2016-02-13:
No, the Pi runs Moonlight AND the VPN client at the same time. The Pi connects to my VPN on boot, and then I run Moonlight with an EmulationStation shortcut. It's all built into one machine.

amivit, 2016-02-13:
Ah so you can bring the pie with you wherever for a portable gaming console? Very nice. I think this combination of retropie, moonlight and kodi is amazing and should get some more attention.

djrockgt, 2016-02-13:
Exactly! I already tried it outside of my home network and it works beautifully. All you need is about 2mbps upload from your home network to run 720p 30fps so don't feel like you need crazy good network speeds. I play Skyrim on LTE using Moonlight on my phone. Lag is minimal.

I agree this needs more attention. It was very hard to do the research for this project. The problem is that Moonlight is a very niche product. You need a gaming rig that can handle it, an nVidia card, and knowledge of Linux terminal because a lot of the steps are still manual. Now that Android set top devices are less than $100 and you can easily install Moonlight and Kodi on them, the need to build the device yourself becomes less attractive. What I would love to see though is a cheap, ultra portable moonlight stick, like the Chromecast, and a Pi image that is almost fully preconfigured. Also if we can somehow find a way to set up the remainder of the settings from the ES menu, THEN we might have a contender.

I'd love to help make this happen. I'm very limited in my Linux knowledge but I learn quick. I built my Pi in a week and I didn't even know what wget was when I started.

tynor88, 2016-02-21:
@djrockgt Sounds awesome. Would be nice if you could provide some scripts or some basic setup guide, for this to run. I'd like to try it out, but i'm no linux expert. :+1:

djrockgt, 2016-02-21:
I'm no expert either and I'm moving my efforts over to OSMC. I don't like the prior RetroPie/Emulationstation build - I don't use either of them except for as a launcher for Moonlight. OSMC just allows me to launch moonlight from Kodi and has a really nice community for add-on development. The Luna addon let's you launch moonlight, pair, etc all from a user interface.

Here's where you can find everything you need: https://discourse.osmc.tv/t/howto-moonlight-embedded-former-limelight/5848

This how-to is very user friendly. The last pieces that I want to figure out is VPN but because it's Debian, my last efforts were completely shot. Once I can contribute something different from what everyone's working on, I'll make my own guide. Right now I'm just reinventing the wheel.

diogomsantos, 2016-02-21:
That is just an addon for Kodi, it should work at least on any Debian based system with Kodi for the Raspberry Pi.
VPN is off-topic but there is a lot of documentation to make a connection on Linux based systems.

RetroPieNerd, 2016-04-15:
If you don't have a compatible Nvidia Graphics card or just don't have one, use this tool to emulate one!

RetroPieNerd, 2016-04-15:
If you don't have a compatible Nvidia Graphics card or just don't have one, use this tool to emulate one!

[nvemulate.zip](https://github.com/RetroPie/RetroPie-Setup/files/221911/nvemulate.zip)

The download is broken on the website, but here is the website anyway: https://developer.nvidia.com/nvemulate

BTW I only managed to download this via the Internet Archive but it is the latest version anyways....

RetroPieNerd, 2016-04-15:
Also, this is official just so you know...
:smile:

ssphaneuf, 2016-04-20:
I just installed retropie for the 1st time on my pi 3 with version 3.7 I've tried it seems there is a package broken somewhere down the line with the source. I can't seem to install moonlight-embedded. Any chance there is an updated version of how to install moonlight embedded or can someone walk me through what i've done werong. i had it installed before i updated to 3.7. Thanks -Scott

BillyWhiz, 2016-04-30:
I got my pi3 to work with 3.7 by making a couple of changes:
Step 3 - change this line to
deb http://archive.itimmer.nl/raspbian/moonlight jessie main
The path to the Emulation Station config file is different, try
sudo nano /home/pi/.emulationstation/es_systems.cfg
The rest of the setup is the same and now the steam logo shows on the menu. Thanks Haris1977

sirus887, 2016-09-28:
@Haris1977 Hi, I am having an a few issues

1. Steam doesnt show in my retropie menu despite its in the config.
2. If I try running the .sh file from command line, I am getting a message saying I must pair the devices. However, if I run the moonlight string as it is from the command line it works fine.

Any ideas?

My setup is:
retropie v4.0.2
Moonlight Embedded 2.2.0
GFE 3.0.6.49

Strange this is, I had this working on a build I done a day ago. :(

Thank you.

Haris1977, 2016-09-29:
@sirus887 sorry but i have abandoned this project, as i moved along steam's **steam link hardware**. I guess my scripts are outdated now. Maybe you can get some help in here

https://github.com/irtimmer/moonlight-embedded

Good luck!

sirus887, 2016-09-29:
Thanks I got it too work,

Tedious, but I started all over with a fresh retropie and done the steps you provided before doing anything else. Maybe something I installed was causing the issue, who knows.

radiske, 2016-11-23:
@Haris1977  thank you for your tutorial, it works great!

However, i had to change

deb http://archive.itimmer.nl/raspbian/moonlight wheezy main

to 

deb http://archive.itimmer.nl/raspbian/moonlight jessie main

:D

Haris1977, 2016-12-01:
Greetings. Even though i have abandoned this here are some changes to my tutorial to get in touch with latest settings:

- Changed the path to the Emulation Station config file to /home/pi/.emulationstation/es_systems.cfg
- Changed location of deb file to deb http://archive.itimmer.nl/raspbian/moonlight jessie main

Thanks @BillyWhiz and @radiske

jdunne525, 2017-01-21:
The procedure didn't work for me.. It seems emulationstation has rearranged the folder structure a bit.  es_systems.cfg is now here:
/etc/emulationstation/es_systems.cfg

I had to run this command to get the es_systems.cfg file in the right spot:
cp /etc/emulationstation/es_systems.cfg ~/.emulationstation/es_systems.cfg

Then continue with the procedure as is... 
sudo nano /home/pi/.emulationstation/es_systems.cfg

Haris1977, 2017-02-07:
Thanks @jdunne525 i 've add it to my how to:)

sysworx, 2017-02-24:
uh were i can get that controller map? My Xbox Controller is normally connected at the host pc. This worked on my macbook to stream steam games and the Xbox Controller works...but for some reason with moonlight the controller works in the Steam Menu but on the game not? 

so i want to try to connect it to the pi, but dont know about that mapping?

chidmdxx, 2017-06-09:
@sysworx don't know if you found a fix for that, but you need to remove/rename some dlls 
https://github.com/moonlight-stream/moonlight-docs/wiki/Setup-Guide#using-a-gamepad-connected-to-the-pc-instead-of-the-streaming-device

BackSlasher, 2017-12-01:
I collected some knowledge to a script and added it to my fork
https://github.com/BackSlasher/RetroPie-Setup
Note sure it's production-ready yet, but would like to get your input

closed 2018-04-23 by Haris1977