RetroPie/RetroPie-Setup #1083: Supertux.sh crashes on exit

Haris1977, 2015-11-13:
As the title says Supertux (installed from experimental packages) crashes on exit.

rpi1 model b, 512 MB, retropie-v3.2.1-rpi1 image

HerbFargus, 2015-11-19:
When you say crash what do you mean by that? Does it freeze?

Haris1977, 2015-11-19:
Only a "black screen" and stays there forever. Have to unplug the power cable and plug it back

HerbFargus, 2015-11-19:
It would be safer to ssh in and reboot than to unplug the cord. https://github.com/RetroPie/RetroPie-Setup/wiki/ssh.

Haris1977, 2015-11-19:
Apart from the unplug thing, is it normal to have a black screen ??(does not revert back to es).

HerbFargus, 2015-11-19:
No its not normal- hence why its in experimental. I'll look into it when I find some time

Haris1977, 2015-11-19:
Ok thanks:)

HerbFargus, 2015-11-20:
did you update your setup script first before installing? the ports function was recently changed and that may have fixed the issue as I just ran it and it did not freeze on exit.

HerbFargus, 2015-11-20:
or if you don't want to reinstall (as the raspbian is offline right now) you can edit the `/home/pi/RetroPie/roms/ports/SuperTux.sh`
to look like this:
```
#!/bin/bash
"/opt/retropie/supplementary/runcommand/runcommand.sh" 0 _SYS_ supertux ""
```

Haris1977, 2015-11-20:
I updated the setup script and re-downloaded the game. Now it works as expected and does not freeze on exit. 

Only drawback is that my left analog stick (wireless xbox compatible controller) is very sensitive to movement.

HerbFargus, 2015-11-20:
glad its sorted.

For the analog settings, You may have to find the deadzone settings in the configuration files in the backend somewhere:

http://supertux.lethargik.org/bugs/view.php?id=169

otherwise you could just set the controls to the dpad,

Haris1977, 2015-11-20:
I find (joystick-deadzone  4096). So what is the number i have to use? lower or higher?

closed 2015-11-20 by Haris1977